TheThingsNetwork/lorawan-stack #6218: Investigate Gateway Server long schedule anytime delays

adriansmares, 2023-05-14:
### Summary

We are sometimes experiencing very long transmission delays when class C downlinks are scheduled without an absolute time (i.e. when a class C downlink is scheduled). We should investigate why this happens, as generally the numbers are absurd (days, to months and sometimes even years), while the maximum delay should be a little over an hour (that's the duty cycle limit).

### Steps to Reproduce

Unknown at this time.

### Current Result

Delays of days to months.

### Expected Result

Delays of at most one hour, and even those should be extremely rare. 1 hour delay means that the GS decided to schedule in the next possible window.

### Relevant Logs

```shell
N/A.
```

### Deployment

The Things Stack Cloud

### The Things Stack Version

3.25.1

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2023-05-19:
> I start to wonder if allowing the delay to grow indefinitely is a good strategy. What happens now is that a downlink path may consume all of the downlinks of multiple end devices and grow basically indefinitely, even if alternatives exist. The fact that we could, by design, grow to delays of a day because the signal is really good from a singular gateway seems slightly bogus to me.

Yes indeed, we should definitely return a "resource exhausted" error from GS if scheduling would happen too far in the future. I would find it acceptable to make that part of closing this issue. I think we can conclude that there are valid reasons to schedule very far into the future, even if it is unlikely. BTW, I'm not surprised that the reasons are "unlikely" as we see this behavior not that often, right?

adriansmares, 2023-05-22:
> BTW, I'm not surprised that the reasons are "unlikely" as we see this behavior not that often, right?

That's right - it seems to be very rare.

adriansmares, 2023-06-27:
A symptom of this issue, based on the logs, is that the gateways periodically reconnect. We observe that these very long scheduling delays occur after the gateway has reconnected.

When a gateway reconnects, the concentrator timestamp synchronization is erased. 

It is also the case that class C downlinks are done generally via the uplink tokens of old uplinks sent by the end device. These uplink tokens contain old timestamps which are most likely too stale to be used for scheduling, but currently the GS will just use these timestamps in the absence of any other synchronization:
https://github.com/TheThingsNetwork/lorawan-stack/blob/1c4220770c7889eab39b70c0281d19dce4a2ad85/pkg/gatewayserver/scheduling/scheduler.go#L256-L268

This mechanism of transferring timestamps via the uplink token exists in order to facilitate situations in which the uplink flow is running on another Gateway Server instance from the instance which has claimed the downlink flow. The problem is that it is hard to establish if the timestamps themselves are stale or not, or even if they are valid (there is no signature/encryption mechanism inside a cluster - we only encrypt the tokens that pass through roaming).

These timestamps (the 32 bit concentrator timestamp, and the 64 bit concentrator time which accounts for rollovers) are valid only within a single 'session' - if the gateway reconnects, or restarts, the old timestamps are invalid and irrelevant. 

Time boxing the tokens is not really water tight, and imposes more limitations than it solves. At the same time, we shouldn't do extra networking on the hot path of the GS (storing the clock state on uplink, and retrieving it on downlink). A more resilient solution is to store the clock information into a shared cache, along with a subscription system. 

We can save + publish the current clock state to a cache, asynchronously, and other instances will retrieve + subscribe when the gateway connects.

---

Implementation details:

1. Remove the `server_time`, `concentrator_time` and `gateway_time` from the `UplinkToken` message.
2. Create a `GatewayClockPubSub` with the following interface:
```go
// GatewayClockPubSub allows PubSub access to the gateway concentrator clock updates.
type GatewayClockPubSub interface {
	// Publish stores the latest known clock state and publishes it to subscribers.
	Publish(
		ctx context.Context,
		ids *ttnpb.GatewayIdentifiers,
		timestamp uint32,
		concentratorTime scheduling.ConcentratorTime,
		serverTime time.Time,
		gatewayTime *time.Time,
	) error
	// Subscribe retrieves the last known clock state and subscribes to state updates.
	Subscribe(
		ctx context.Context,
		ids *ttnpb.GatewayIdentifiers,
		f func(
			ctx context.Context,
			timestamp uint32,
			concentratorTime scheduling.ConcentratorTime,
			serverTime time.Time,
			gatewayTime *time.Time,
		) error,
	) error
}
```
3. Add a new channel and background task, similar to how the gateway connection stats works, which stores and publishes the clock state. Updates should be published only if the uplink stream is active and the downlink stream is inactive (to avoid loops).
4. When the gateway connects, another background task retrieves the last known state from the store and subscribes to updates. Updates should be subscribed to only if the downlink stream is active and the uplink stream is inactive (to avoid loops).

Only the UDP frontend needs to participate in this protocol. The stored last known state needs to have a TTL attached to it, in order to avoid storing the state forever. 

---

Any thoughts and/or comments on this solution ?

johanstokking, 2023-06-28:
Sounds good.

KrishnaIyer, 2023-06-28:
Proposal looks good. From how I read this, no two instance will share this state right? The instance with the "connection"(upstream socket) writes to cache when disconnecting and the instance to which the gateway next "connects" to reads the sate.
If this understanding is right, I prefer the `Save/Restore` semantics instead of `Pub/Sub` since the later seems to indicate concurrent subscribers.