TheThingsNetwork/lorawan-stack #6322: Specify who is creating a new API Key

nicholaspcr, 2023-06-12:
### Summary

The email template for API_KEY creation informs about the creation of said key but it does not provide who is the entity responsible for its creation, leaving the email which should provide context to admins without an important information.

### Current Situation

What is described on the summary, emails don't inform who is creating the api_key.

### Why do we need this? Who uses it, and when?

So that the email can provide a more detailed information regarding api_key creations.

### Proposed Implementation

Requirements:
- Add the `AuthInfoResponse.GetEntityIdentifiers()` into the `api_key_create` template.
  - Probably interesting to have the same in the `api_key_change` template as well.
- Add tests for the template information.

### Contributing

- [ ] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-06-13:
I don't think this is a good idea. I can see why this info is needed in the notification but I'm concerned about potential future bugs leading to privacy breaches. See https://github.com/TheThingsIndustries/lorawan-stack/issues/3384

KrishnaIyer, 2023-09-27:
This item will be a part of broader audit logging. I'll take it up then.

closed 2023-09-27 by KrishnaIyer