TheThingsNetwork/lorawan-stack #6337: Register end device is reporting: No DevEUI specified when generating or entering manually

jrepp, 2023-06-22:
### Summary

I am attempting to register a new device manually. This process worked last week, but now the form doesn't accept the DevEUI. I have tried manually entering a DevEUI and generating one. Both attempts failed.

This is all random data but you can see a sample registration
[image]

[image]

### Steps to Reproduce

Register a device manually.
Use a all zeros app ID or generate a unique one
Enter a unique DevEUI or generate one with the form. Note that 1/X is incremented.
Fill in all form fields.
Register new device

Fails with:

`{
  "code": 3,
  "message": "error:pkg/joinserver:no_dev_eui (no DevEUI specified)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/joinserver",
      "name": "no_dev_eui",
      "message_format": "no DevEUI specified",
      "correlation_id": "c7917c1f74d04be2a580ceff65628bfd",
      "code": 3
    }
  ]
}`

### Current Result

An error occurs that the DevEUI is not specified

`{
  "code": 3,
  "message": "error:pkg/joinserver:no_dev_eui (no DevEUI specified)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/joinserver",
      "name": "no_dev_eui",
      "message_format": "no DevEUI specified",
      "correlation_id": "c7917c1f74d04be2a580ceff65628bfd",
      "code": 3
    }
  ]
}`

### Expected Result

The device should be registered.

This worked adequately last week with multiple devices.

### Relevant Logs

```shell
{
  "code": 3,
  "message": "error:pkg/joinserver:no_dev_eui (no DevEUI specified)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/joinserver",
      "name": "no_dev_eui",
      "message_format": "no DevEUI specified",
      "correlation_id": "c7917c1f74d04be2a580ceff65628bfd",
      "code": 3
    }
  ]
}
```

### URL

https://eu1.cloud.thethings.network/console/applications/css-facility-poc/devices/add

### Deployment

The Things Stack Community Edition

### The Things Stack Version

3.26.1

### Other Information

Worked last week.

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2023-06-23:
@jrepp this has been fixed. This is a regression in https://github.com/TheThingsNetwork/lorawan-stack/pull/6335 that was unfortunately already deployed.

closed 2023-06-23 by johanstokking

jrepp, 2023-06-24:
Thank you @johanstokking!