TheThingsNetwork/lorawan-stack #6345: Deprecate generate-qr command for end devices

KrishnaIyer, 2023-06-30:
### Summary

Deprecate `generate-qr` command for end devices. Refs 
https://github.com/TheThingsIndustries/lorawan-join-server/issues/55 and https://github.com/TheThingsIndustries/product-management/issues/13

### Current Situation

The `generate-qr` command requires an end device to be registered to generate the QR Code. This doesn't work with the new claiming flow.

We should also maybe deprecate the `Generate` function of the `EndDeviceQRCodeGenerator` since QR generation should be a function of an external join server.

### Why do we need this? Who uses it, and when?

This is to remove functionality that doesn't work.

### Proposed Implementation

Follow the usual deprecation steps.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2023-06-30:
So what do we do, return an error in `generate-qr`?

KrishnaIyer, 2023-07-03:
I think we should do a `noop` with a warning that asks the user to use `ttjs-cli` and point to the docs.

johanstokking, 2023-07-03:
I'm not sure what you mean with `noop`. This command is expected to write an image to a file. If we suddenly stop doing that while returning `0` exit code, we break CLI compatibility.

This command is marked `EXPERIMENTAL`, but still, changing it to not doing anything breaks scripts that are using it.

I don't see another way of returning an error.

KrishnaIyer, 2023-07-03:
Ok I see. Then let's indeed return an error with a message to use the `ttjs-cli` and point to the docs.

PR #6370 (2023-07-04): Remove QR code functionality from CLI

#### Summary

Closes #6345

#### Changes

- Remove QR code format listing and generation from CLI and hide the commands. These commands were marked EXPERIMENTAL so we can do anything with them
- Hide `provision` command; it still works and we can't remove it as it wasn't EXPERIMENTAL

#### Testing

N/A

##### Regressions

Users and scripts depending on this command will stop working, but because of the way they were marked, the compatibility commitment does not apply.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated  [... cut]
commit c0281f14 (2023-07-04): cli: Remove QR code generation functionality

Closes #6345
closed 2023-07-04 by johanstokking