TheThingsNetwork/lorawan-stack #6376: Remove all class components and replace them with hooks

kschiffer, 2023-07-05:
### Summary

To follow the latest best practices and simplify implementations, we should work to replace any React class component with a functional component. We already converted many in the last years and I replaced many more as part of #6327 but there is still a substantial amount left.

### Current Situation

Many older components still use the class-based approach, which is becoming increasingly cumbersome as it often causes incompatibilities with upgraded dependencies, such as react-router v6 but also overall constitutes a code smell. Latest react versions clearly state functional components as the preferred modus.

### Why do we need this? Who uses it, and when?

We need this to address technical debt and to make our codebase more future-proof. It's also annoying to have to convert classes on the spot while working on other features, as it takes additional time, poses potential error risks and dilutes the diff which makes reviewing harder.

### Proposed Implementation

Let's make sure we do this in a consistent manner. Here's are the guidelines I followed when converting components:
1. Class components that make use of `withRequest()`-HOC  to fetch initial data should be split into two functional components (outer and inner, arrow functions, the outer should keep the original name), so that \<RequireRequest requestAction={exampleActionCreator(param1, param2)} /> can be run in the outer function to ensure that all required data will be available in the inner class when using `useSelect`.
3. Class components that make use of `connect()`-HOC should be refactored to use `useSelect()` instead (in the inner component)
4. Class components that make use of `withBreadcrumbs()`-HOC should use `useBreadcrumbs('id.path', <Breadcrumbs … />)`-hook instead (same signature as `withBreadcrumbs`)
5. Class components that make use of `withFeatureRequirement()`-HOC for feature flags should use `<Require featureCheck={mayDoAction} otherwise={{ redirect: '/example-route' }} />` instead
6. Classes using Routes should use relative paths (react router v6)
7. All hooks and handlers should be wrapped in `useCallback()`
8. Unused proptypes should be removed

### Contributing

- [ ] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [ ] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

kschiffer, 2023-07-05:
Once we have all class components removed, we'll also be able to refactor our redux-actions to get rid of `REQUEST|SUCCESS|FAILURE` actions, as we can fully switch to promisified dispatches. This will make action definition a lot simpler, readable, and maintainable.

KrishnaIyer, 2023-07-12:
As discussed offline, @ryaplots will take a crack at this this milestone to see how far we can get.

kschiffer, 2023-07-24:
We're also quite far already with removing all `withRequest` instances.

Last standing are:
- [ ] ./pkg/webui/console/containers/application-title-section/connect.js
- [ ] ./pkg/webui/console/containers/gateway-title-section/connect.js
- [ ] ./pkg/webui/console/containers/organization-title-section/connect.js

Once we've refactored these, we can remove `withRequest` entirely and with it all `fetching`, `error` store logic.

ryaplots, 2023-08-03:
We also have some functional components that still use `connect(`:
- [ ] ./pkg/webui/lib/components/full-view-error/connect.js
- [ ] ./pkg/webui/lib/components/full-view-error/index.js
- [ ] ./pkg/webui/account/containers/collaborators-table/index.js
- [ ] ./pkg/webui/console/components/uplink-form/connect.js
- [ ] ./pkg/webui/console/components/uplink-form/index.js
- [ ] ./pkg/webui/console/components/downlink-form/connect.js
- [ ] ./pkg/webui/console/components/downlink-form/index.js
- [ ] ./pkg/webui/console/containers/gateway-location-form/connect.js
- [ ] ./pkg/webui/console/containers/gateway-location-form/index.js
- [ ] ./pkg/webui/console/containers/application-title-section/connect.js
- [ ] ./pkg/webui/console/containers/application-title-section/index.js
- [ ] ./pkg/webui/console/containers/application-events/index.js
- [ ] ./pkg/webui/console/containers/gateway-title-section/connect.js
- [ ] ./pkg/webui/console/containers/gateway-title-section/index.js
- [ ] ./pkg/webui/console/containers/device-profile-section/device-selection/brand-select/connect.js
- [ ] ./pkg/webui/console/containers/device-profile-section/device-selection/brand-select/index.js
- [ ] ./pkg/webui/console/containers/device-profile-section/device-selection/model-select/connect.js
- [ ] ./pkg/webui/console/containers/device-profile-section/device-selection/model-select/index.js
- [ ] ./pkg/webui/console/containers/device-profile-section/device-selection/fw-version-select/connect.js
- [ ] ./pkg/webui/console/containers/device-profile-section/device-selection/fw-version-select/index.js
- [ ] ./pkg/webui/console/containers/device-profile-section/device-selection/hw-version-select/connect.js
- [ ] ./pkg/webui/console/containers/device-profile-section/device-selection/hw-version-select/index.js
- [ ] ./pkg/webui/console/containers/device-profile-section/device-card/connect.js
- [ ] ./pkg/webui/console/containers/device-profile-section/device-card/index.js
- [ ] ./pkg/webui/console/containers/applications-table/index.js
- [ ] ./pkg/webui/console/containers/gateway-connection/connect.js
- [ ] ./pkg/webui/console/containers/gateway-connection/index.js
- [ ] ./pkg/webui/console/containers/device-events/index.js
- [ ] ./pkg/webui/console/containers/device-title-section/connect.js
- [ ] ./pkg/webui/console/containers/device-title-section/index.js
- [ ] ./pkg/webui/console/containers/organization-events/connect.js
- [ ] ./pkg/webui/console/containers/organization-events/index.js
- [ ] ./pkg/webui/console/containers/dev-addr-input/index.js
- [ ] ./pkg/webui/console/containers/organization-title-section/connect.js
- [ ] ./pkg/webui/console/containers/organization-title-section/index.js
- [ ] ./pkg/webui/console/containers/owners-select/connect.js
- [ ] ./pkg/webui/console/containers/owners-select/index.js
- [ ] ./pkg/webui/console/containers/gateway-events/index.js

KrishnaIyer, 2023-08-16:
Rolling item in progress. Moved to `v3.27.2`

ryaplots, 2023-10-16:
Referencing [this comment](https://github.com/TheThingsNetwork/lorawan-stack/pull/6621#discussion_r1360506432).
`Checkbox group` and `radio button group` need a proper refactoring.

kschiffer, 2023-12-14:
Closing this now as completed. The remaining two components can be refactored later. What's important now is that the vast majority of components are refactored and we can now look into some more general restructuring of our codebase [such as refactoring fetching and error logic](https://github.com/TheThingsNetwork/lorawan-stack/issues/6376#issuecomment-1621460407) which is now done completely based on promisified dispatches rather than store artifacts.

Thanks @ryaplots and @PavelJankoski for the hard work on this. This made our codebase a whole lot more future-proof!

closed 2023-12-14 by kschiffer