TheThingsNetwork/lorawan-stack #6441: Webpage expanding horizonal menu for structure swapped for end devices

jezdd, 2023-08-02:
### Summary

Menu expands but end devices link gets swapped with the device name...?

[image]

### Steps to Reproduce

Open applications, open end devices, open any device, check horizontal menu structure. 

### Current Result

Horizontal menu structure back to from on the last two items  

### Expected Result

Applications > Apiary Sensors > End devices > "device name"

### Deployment

The Things Stack Enterprise (self-hosted)

### The Things Stack Version

3.27.0

### Client Name and Version

```shell
Chrome browser
```

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-08-07:
Yup this is valid. @ryaplots: Is this a result of how we setup the breadcrumb thing?

ryaplots, 2023-08-10:
> Yup this is valid. @ryaplots: Is this a result of how we setup the breadcrumb thing?

Yeah, indeed. It uses the render order, so maybe at some point the render order does not correspond to the order of the actual components. I'll need to debug

PR #6468 (2023-08-18): Fix order of breadcrumbs for device pages in the Console
#### Summary
Closes #6441

#### Changes

- Fix the breadcrumb IDs so that they will be ordered correctly

#### Testing

- Navigate to all device subpages and check whether the breadcrumbs are displayed in the right order.

#### Notes for Reviewers

I've recently changed the logic for how the breadcrumbs determine their order from formerly being determined by render order, to determine the order by the breadcrumb IDs (alphabetically). This is better because the render order does not always correspond to the navigation structure.
To make t [... cut]
closed 2023-08-18 by kschiffer