TheThingsNetwork/lorawan-stack #6469: Remove "could not" and "failed to" from error messages

KrishnaIyer, 2023-08-17:
### Summary

Remove "could not" and "failed to" from error messages. An error message implies that an action could not be completed or it failed.

We have a list of legacy errors that still have these and let's remove these and make @johanstokking happy.

### Current Situation

There's a collection of older error definitions that start with `could not` or `failed to`.
Examples
```go
errEndDeviceClaimInfo = errors.DefineFailedPrecondition(
"end_device_claim_info",
"could not get end device claim info from DCS",
)
errPkgDataMerge     = errors.DefineCorruption("pkg_data_merge", "failed to merge package data")
``` 

### Why do we need this? Who uses it, and when?

To make the codebase consistent.

### Proposed Implementation

1. Do a search in `*.go` files for `could not` and `failed to`.
2. If the message is part of an error definition ex: `errXYZ = errors.DefineXYZ ("xxx", "could not --")` remove these words and.
3. Update the error messages to make grammatical sense. Ex; `could not connect to Join Server` -> `connect join server`. Since this text is part of an error message, it's grammar make sense.
4. Regenerate messages `tools/bin/mage go:messages.`
5. Repeat for enterprise.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2023-08-17:
It's gonna help a lot to look in `messages.json` probably

johanstokking, 2023-08-31:
The vast majority of error messages follow this wording already. The error message basically says "what" went wrong. Indeed, for display, it needs to be clear there was an error. Best case, the action or desired outcome is known (i.e. "list applications failed") and the error message refers to the actual thing that failed.

kschiffer, 2023-09-01:
I believe we have to remove any displayed backend error from the frontend and replace it with a generic error message for the time being to avoid showing incomplete error messages then.

To be honest, I don't get the practical use of cutting off error messages like that, other than convenience for ourselves when writing them. If we regard the error messages as potentially end-user-facing text, which I believe we do, then in my opinion it makes no sense to obfuscate the meaning like that.

johanstokking, 2023-09-01:
> I believe we have to remove any displayed backend error from the frontend and replace it with a generic error message for the time being to avoid showing incomplete error messages then.

Showing it as the only message is incomplete indeed. It should be clear there's an error, and I think the error code and message serve fine as a cause. Having only a generic message and nothing else makes support really hard.
 
> To be honest, I don't get the practical use of cutting off error messages like that, other than convenience for ourselves when writing them. If we regard the error messages as potentially [...]

It's not the convenience even. First, it's more consistent with runtime errors. Second, errors can be concatenated by wrapping (`fmt.Errorf("dial %q: %w", addr, err)`) so having "failed to" and "could not" with a string of causes gets long (although this works slightly different in TTS with our cause hierarchy). Third, related to that, error messages are not full sentences starting with a capital and punctuation. So it'll never be `Failed to dial target "xyz".`: runtime (and many library) errors are not like this, wrapping won't work and error messages should be lowercase without punctuation.
 
> Otherwise, we may want to create an additional mapping that maps errors to user-friendly text for the Console/CLI, but that would be very tedious.

Yeah I agree with this. Though the Console is a presentation layer, it's gotta figure out how to display errors. Making full sentences is just not the backend's job as the user intention isn't known so there's a lot of context missing.

I don't see the problem with displaying: `X failed. Cause: %s`; the Console knows `X` because it's the frontend action, like buttons or other UX things, and `%s` is just the error message.

PR #6484 (2023-09-05): Remove "could not" and "failed to" from error messages

#### Summary

Closes #6469 

#### Changes

- Remove "could not" and "failed to" from error messages

#### Testing

CI

##### Regressions

n/a

#### Notes for Reviewers

To find files with error definitions containing "could not" and "failed to" I've created a simple script:
```bash
jq -r <config/messages.json \
  '.[]
  | select(
    .translations.en
    | contains("could not") or contains("failed to")
  )
  | .description
  | "\(.
closed 2023-09-05 by happyRip