TheThingsNetwork/lorawan-stack #6483: Come up with way to store user specific preferences for use in the Console

kschiffer, 2023-08-28:
### Summary

As part of the ongoing UX improvement plans for the Console, it will be beneficial to store certain user preferences in the user record so they can be accessed universally.

### Current Situation

Right now, it is only possible to store attributes to entities (including users). Using this interface is only good for storing strings. This means that while we could store some data as simple strings and/or JSON, this approach has a couple of drawbacks:
- This is using attributes (supposedly) against their original purpose
- While simple strings can be indexed and searched, it's not possible with JSON formatted data
- Not easy to mark such special attributes so that they don't interfere with the other uses of attributes

I believe this why we should think of a new way to store such data.

### Why do we need this? Who uses it, and when?

We need this for purposes such as:
- Storing bookmarks/favorite entities so they can be accessed more easily when navigating the Console
- Storing user preferences such as: dark mode, customized dashboard layouts, different list view layouts, default sort orders, default event filters, etc
- Storing payload formatters for different purposes to apply them easily when onboarding different end devices

There are various other ways we could make use of such user data.

### Proposed Implementation

First of all, I want to point out that while using browser-based storage such as cookies/local storage can be used to address this issue as well, the biggest drawback is that such preferences would be attached to the browser and as such lost whenever a user logs in from a different browser (eg when using mobile). This is less dramatic for some settings such as preferred sort orders, but would become more frustrating when storing bookmarks such as favorite end devices or gateways.

As such, I see two remaining options:

### Backend-for-frontend

This approach would mean that we set up a small service (eg. express.js server) that would connect to a database to store such user-defined data and generally act as a middleware between Console and backend.

**Pros:**
- Quite flexible
- Could be used to do other things such as acting as middleware for connections to the backend and providing a simple interface to facilitate the tasks of the Console

**Cons:**
- Lots of added complexity (as in failure potential) and extra maintenance
- There would need to be a mechanism that keeps the two databases reasonably in sync, eg. removing bookmarks for deleted entities
- Performance overhead, as the added complexity and fetching steps can slow down the overall experience

### Extend user registry

This is I guess the most intuitive approach, where we would add new fields to the user schema that are then handled exclusively by the Console to store user preferences.
**Pros:**
- Relatively simple to implement and maintain
- Possible to query for certain data

**Cons:**
- Storing and maintaining data that is exclusively used by the Console
- There might be limits on how much the schema can be modified and expanded before it starts to impact performance

I myself would lean to the latter approach but I might not understand all possible concerns, so I'd like to gather some opinions here so we can decide which approach to take.

cc @adriansmares @johanstokking @KrishnaIyer 

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

nicholaspcr, 2023-08-30:
I'm not so sure if its ideal to store the console user preferences on the Identity Server as it is a set of information which aren't attached to entities but rather to desired user behaviors on the front-end. Having said that, I still agree that its probably the best choice to store said information on the IS purely due to the decrease of complexity that this approach brings.

I also agree with the two approaches of storage given by @adriansmares.

kschiffer, 2023-08-31:
Great, then let's go with the IS approach.

> These are simply columns per user, right?

Let's see:
- Bookmarks: many-to-many table, or array column
- Dark mode: boolean in the user record
- Dashboard layout: this can potentially be more complex, since it should store panel ID, order and possibly a configuration; is it a good idea to store this as a JSON?
- Sort orders: this should ideally be saved per table, so I guess this would also be a JSON in the user table?
- Payload formatters: let's skip this for now, it was just a first idea

@KrishnaIyer can you help to schedule this in conjunction with the Console design updates?

KrishnaIyer, 2023-09-01:
Yes. This is added as an item to https://github.com/TheThingsIndustries/product-management/issues/29.

KrishnaIyer, 2023-12-18:
@nicholaspcr: Do you have sufficient input in the issue to start implementing this?

nicholaspcr, 2023-12-18:
@KrishnaIyer yeah the suggestion made by Adrian is good solution. Will work on this after finishing the `contact_info` removal issues.

nicholaspcr, 2024-03-06:
Updating this issue with the changes made to the IS, this should unblock the front-end issues related to these features. 
Tagging @kschiffer so that the Console changes can be distributed to the front-end team.

<hr/>

Changes introduced by https://github.com/TheThingsNetwork/lorawan-stack/pull/6890 

A new field `console_preferences` was introduced to user proto message:
```json
{
    "console_theme": "('CONSOLE_THEME_SYSTEM'|'CONSOLE_THEME_LIGHT'|'CONSOLE_THEME_DARK')",
    "dashboard_layouts": {
        "api_key":      "('DASHBOARD_LAYOUT_TABLE', 'DASHBOARD_LAYOUT_LIST', 'DASHBOARD_LAYOUT_GRID')",
        "application":  "('DASHBOARD_LAYOUT_TABLE', 'DASHBOARD_LAYOUT_LIST', 'DASHBOARD_LAYOUT_GRID')",
        "collaborator": "('DASHBOARD_LAYOUT_TABLE', 'DASHBOARD_LAYOUT_LIST', 'DASHBOARD_LAYOUT_GRID')",
        "end_device":   "('DASHBOARD_LAYOUT_TABLE', 'DASHBOARD_LAYOUT_LIST', 'DASHBOARD_LAYOUT_GRID')",
        "gateway":      "('DASHBOARD_LAYOUT_TABLE', 'DASHBOARD_LAYOUT_LIST', 'DASHBOARD_LAYOUT_GRID')",
        "organization": "('DASHBOARD_LAYOUT_TABLE', 'DASHBOARD_LAYOUT_LIST', 'DASHBOARD_LAYOUT_GRID')",
        "overview":     "('DASHBOARD_LAYOUT_TABLE', 'DASHBOARD_LAYOUT_LIST', 'DASHBOARD_LAYOUT_GRID')",
        "user":         "('DASHBOARD_LAYOUT_TABLE', 'DASHBOARD_LAYOUT_LIST', 'DASHBOARD_LAYOUT_GRID')"
    },
    "sort_by": {
        "api_key":       "same fields allowed in the respective list entity request",
        "application":   "same fields allowed in the respective list entity request",
        "collaborator":  "same fields allowed in the respective list entity request",
        "end_device":    "same fields allowed in the respective list entity request",
        "gateway":       "same fields allowed in the respective list entity request",
        "organization":  "same fields allowed in the respective list entity request",
        "user":          "same fields allowed in the respective list entity request"
    }
}
```

The idea here is that the front-end team can add new fields to the `console_preferences` proto definition without
requiring a migration, since the bytes of the object is what is stored on the database. So this field is malleable
enough for new fields at any point.

<hr/>

Changes introduced by https://github.com/TheThingsNetwork/lorawan-stack/pull/6968

A new registry was added `UserBookmarkRegistry`.

https://github.com/TheThingsNetwork/lorawan-stack/blob/282b6a3176424551e4f525cc6a8f2781ff1a120b/api/ttn/lorawan/v3/user_services.proto#L176-L208

The operations are quite straight forward. Bookmarks are entities that can be soft-deleted but not by any of the
registry operations, only when an entity referenced is deleted that the bookmarks is soft-deleted. 

So when user-X has a bookmark to entity-Y, if entity-Y is deleted and restored the bookmark will remain present and it
will be returned in the ListBookmarks request.

Additionally there is support for listing deleted bookmarks:
https://github.com/TheThingsNetwork/lorawan-stack/blob/282b6a3176424551e4f525cc6a8f2781ff1a120b/api/ttn/lorawan/v3/user.proto#L554-L555

But I'm not sure if it is wise to show deleted bookmarks, unless requested by users later on.

kschiffer, 2024-03-07:
Thank you! I believe we can close this issue then, since all Console related issues are filed individually.

cc @ryaplots

closed 2024-03-07 by kschiffer