TheThingsNetwork/lorawan-stack #6519: Live Data no longer visible

pessinat, 2023-09-06:
### Summary

We used the stack for about a year and everything worked well. Recently, starting with a PC than another and another the live data in the device and application page are not working anymore.
It's strange that not on all PC doesn't work and that it doesn't stop working at the same time (but one at a time,after weeks).
We tried on a new PC and it works, but after few login with different account, it stopped working.

We tried different browser and also by clearing cache and browser history but nothing change. We tried also to connect the PC via mobile hotspot, to bypass the company firewall, but nothing change.

### Steps to Reproduce

Unfortunately i don't know how you can reproduce it.

### Current Result

In attachment you will find the HAR file and the screenshot of the browser console in the **not** working situation
[KO.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/12534192/KO.zip)

### Expected Result

In attachment you will find the HAR file and the screenshot of the browser console in the working situation
[OK.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/12534193/OK.zip)

### Deployment

The Things Stack Cloud

### The Things Stack Version

TTS v3.27.1

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-09-06:
So the issue is that live data in the end device view is not loading in some cases intermittently?

pessinat, 2023-09-06:
Yes but not intermittently, when on a PC stop working it never work again. Until few days ago we have some PC where it works and other where it does not work. Now it does not work of every our PC. 
The attached screenshot is what I see.
[image]

kschiffer, 2023-09-07:
Hey @pessinat, thanks for creating this issue and sorry to hear you have problems with the live data panel. We've heard about this issue before but we're still not able to reproduce it at all. To help me out in assessing the situation can you answer the following questions:

- What exact chrome version are you using (from the HAR, I can see it's 116.x but I'd like to know the exact version)
- What exact Windows version are you using?
- When the issue happens, do you have any other tabs open that run the Console?
- Are you able to consistently reproduce the issue. Eg, would it occur every time you refresh or only sometimes?

Also, the HAR you kindly provided do not include the relevant stream request. This is because the stream is still active when you generate the HAR. To generate another HAR that includes the stream request, please refresh the live data page, wait for a couple of seconds (15seconds for testing purposes) and then navigate away from the event page (this will cause the stream request to be closed). After you did that, please export the HAR and attach it.

pessinat, 2023-09-07:
Hi @kschiffer thanks your reply. Here the requested information:
- What exact chrome version are you using (from the HAR, I can see it's 116.x but I'd like to know the exact version)
It's the version 116.0.5845.180 
- What exact Windows version are you using?
Windows 10 Enterprise, version 21H2, build 19044.3324
- When the issue happens, do you have any other tabs open that run the Console?
After the problem happens once, it will happen forever no matter how many tabs I have open
- Are you able to consistently reproduce the issue. Eg, would it occur every time you refresh or only sometimes?
Unfortunately, like I said, i can't reproduce it because now it happens forever. The only experimnt that we take is as follow:

1. take a brand new PC
2. log in with an account that doesn't work on the person's ("old"/usual) PC
3. it works
4. we tried to log out and login again -> OK
5. Now we tried with another account (mine) and it doesn't work
6. Now even the other account doesn't work anymore.

Moreover, we tried to login on my PC with the account of another person which doesn't have the problem on his pc. But on my machine it doesn't work. For the requested test we are going to use this "working" PC as control.

I will be back with the file HAR as you asked.

Thanks

pessinat, 2023-09-07:
Here the HAR files. Please, let me know if they are ok. Thanks.
[new har.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/12550604/new.har.zip)

pessinat, 2023-09-07:
Some additional information. The last activity count seems to update only when the page is refreshed.
The usual control device send a message every 30 second and i correctly see them with my parallel MQTT.
here some picture:

I wait some minutes, and in the meantime the counter keep going forward:
[image]

If I refresh the page, the counter updates:
[image]

KrishnaIyer, 2023-09-13:
Thanks for the information and helping us with the debugging. 

We are internally considering technical changes to make the events stream (which is used for the `Live Data` view) more stable. I'll post in this issue once we come up with a decision on how to move forward.

closed 2023-11-23 by PavelJankoski