TheThingsNetwork/lorawan-stack #6529: Switch camera button during device onboarding via mobile doesn't work

KrishnaIyer, 2023-09-08:
### Summary

Switch camera button during device onboarding via mobile doesn't work.
Ref: https://github.com/TheThingsNetwork/lorawan-stack/pull/6500

### Steps to Reproduce

See https://github.com/TheThingsIndustries/lorawan-stack-support/issues/991#issuecomment-1710017134. 
I have also tested this with an iPhone 12 Pro with Safari and Firefox.

### Current Result

The switch camera button doesn't do anything visible on clicks.

### Expected Result

The button should cycle through the cameras. 
Since this is used by users to onboard devices, we need the users to be able to use the rear camera of their mobile phones. 
The good thing is that I have consistently seen that it always shows the primary (rear) camera. 
If we can somehow guarantee that this works, we don't even need to cycle I think.

### Relevant Logs

```shell
See https://github.com/TheThingsIndustries/lorawan-stack-support/issues/991#issuecomment-1710017134
```

### Deployment

The Things Stack Open Source (self-hosted)

### The Things Stack Version

3.27.2

### Client Name and Version

```shell
iPhone Pro with Firefox and Safari.
```

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

ryaplots, 2023-10-19:
Merged a fix which works for android. Although still broken on iPhone.

KrishnaIyer, 2023-11-02:
@mjamescompton: Can you please pick this up since you have expertise in setting up dev environments for iOS?

PR #6691 (2023-11-13): Fix camera switching for ios, android and desktop

#### Summary

Closes #6529

#### Changes

- Refactor Video component and separate into `Camera`, `Stream`, `Video`  components.
- Only display video stream based on `videoinput` with a `deviceID`

#### Testing

Testing
Log in the console on a mobile device
Go to register end device
Click on scan end device QR code
See the switch button at the bottom of the video
Click and see the camera changing

##### Regressions

...

#### Notes for Reviewers