TheThingsNetwork/lorawan-stack #6565: Easier linking to applications in other clusters

eggfriedrice, 2023-09-15:
### Summary

It would be nice if applications list had easier links to other cluster consoles for applications on other clusters

### Current Situation

At the moment you get a list with some greyed out applications names and this "Other cluster" warning:
[image]

### Why do we need this? Who uses it, and when?

Anyone with applications in multiple clusters

### Proposed Implementation

Make the "Other cluster" links clickable to go to the application in that other cluster. It might need a pop up warning or something, but it would be nice to be able to click, especially as the url for the target cluster is shown in the pop up help

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-09-27:
Thanks @eggfriedrice for the issue.

This is possible for applications and for gateways where the LNS address is in a different cluster on the same deployment. We need some Regex to validate this and provide a link.

KrishnaIyer, 2025-04-30:
We'll come up with better ways of doing this moving forward.

closed 2025-04-30 by KrishnaIyer