TheThingsNetwork/lorawan-stack #6571: Class B C gateway is not handled when downlink is sent via CLI

onizmx, 2023-09-21:
### Summary

When class c downlink is sent via
```
ttn-lw-cli end-devices downlink push --application-id halter --device-id 'xxxxx' --f-port 1 --frm-payload 'xxxx' --class-b-c.gateways eui-xxxx--class-b-c.absolute-time '2023-09-11T00:54:51.000000Z'
```
we expect a class c attempt with a give list of gateways.
Confirmed in console that the downlink message has no list of class b c gateways.

### Steps to Reproduce

1. Send class c downlink via `ttn-lw-cli end-devices downlink push --application-id halter --device-id 'xxxxx' --f-port 1 --frm-payload 'xxxx' --class-b-c.gateways eui-xxxx--class-b-c.absolute-time '2023-09-11T00:54:51.000000Z'`
2. Check console for the multicast device and confirm no downlink pass error is returned. CLI should fail too.

### Current Result

CLI returns no downlink path error. Because CLI code is not handling gateway arguments.

### Expected Result

Class C message is attempted for the given list of gateways.

### Relevant Logs

```shell
This is the current fix we've applied
https://github.com/halter-corp/lorawan-stack/commit/8c7a142e07394f2be86392b44f9634ccdefff9a8
```

### Deployment

The Things Stack Open Source (self-hosted)

### The Things Stack Version

3.27.2

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-09-26:
This looks like a bug but I don't have an opportunity to test at the moment. @adriansmares can you check?

adriansmares, 2023-09-26:
This bug is valid, and the current code is indeed unusable. This is probably the result of a very old refactoring that didn't get tested at the time.

closed 2023-09-27 by adriansmares