TheThingsNetwork/lorawan-stack #6573: List deleted entities does not work for non admin users

nicholaspcr, 2023-09-25:
### Summary

While looking into https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/1165#issuecomment-1726416533 I noticed that when attempting to fetch deleted entities we are firstly trying to fetch accounts which trigger the `_isSoftDelete` conditional in the `applyContext` executed in most queries in the store pkg. Since `DeletedOptions`  isdetermined in the registry level and that propagates to the membership section of the store pkg, it means that we are trying to find an deleted account by passing a valid user.

### Steps to Reproduce

List an entity: 
1. Create an app
2. Delete an app
3. Run `ttn-lw-cli app list --deleted`
4. You will get the `account_not_found` error.

Search for an entity:
1. Login into a non admin user
2. Create an app
3. Delete an app
4. Run `ttn-lw-cli app search --deleted`
5. You will get the `account_not_found` error.

### Current Result

We always attempt to search for an account with `deleted_at IS NOT NULL`  but that is not the desired behavior

### Expected Result

Get a valid user and then search for deleted entities/memberships.

### Deployment

The Things Stack Enterprise (self-hosted)

### The Things Stack Version

3.27.2-dev

### Client Name and Version

```shell
Development version
```

### Proposed Fix

1. Update methods in the `membership_store.go` to only search for non deleted users, overwriting the `DeletedOptions` on only the calls for `getAccountModel`.
2. Add tests that validate this.

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-09-26:
I don't understand the requirement. Why should non admins be able to list deleted entities? 
Admins need this to purge IDs.

nicholaspcr, 2023-09-26:
Maybe from me trying to fit in both `list` and `search` under the same behavior, it causes the confusion.

1. `list` fails even for admin users, the reason for that is trying to find the user as a deleted account due to the injection on `DeletedOptions` in the context.
2. `search` only works for admins because fetching the memberships are skipped by setting `member` to nil in the search related methods ([ref](https://github.com/TheThingsNetwork/lorawan-stack/blob/e647970ebc4a96c8dbbf254e02305d7a39f2e586/pkg/identityserver/registry_search.go#L44C2-L46))
3. Even if a user should be forbidden to see its own deleted entities due to the lack of admin privileges, the error should reflect that. Instead of informing that the account was not found.
```
error:pkg/identityserver/store:account_not_found (user account with id `admin` not found)
    account_type=user
    account_id=admin
    correlation_id=8ca168ed8379438cbe6066d5cbcbe6dc
exit status 255
```

In addition of that I would argue that an user being able to see a list of its own deleted entities isn't problematic, one could even check before asking an admin to restore the entity. What you think @KrishnaIyer ?

KrishnaIyer, 2023-09-26:
Before getting into the details of implementation, let's first discuss what we're trying to achieve.

> In addition of that I would argue that an user being able to see a list of its own deleted entities isn't problematic, one could even check before asking an admin to restore the entity

Is this a customer requirement? What is the basis for this? I think that the current flow of only allowing admins to list purged entities works well. Users will get an `ID taken` or similar warning and then they can request an admin to purge.

PR #6576 (2023-10-03): Fix for `WithSoftDelete` failing some specific operations

#### Summary

Closes #6573

The issue revolves around two things:
1. Operations of `search` were failing for non admin users.
2. Operations of `list` were failing for both admin and non admin users.

The reason being that the `WithSoftDeleted` impacts the account search that is done on the membership store, resulting in the following error:
```
error:pkg/identityserver/store:account_not_found (user account with id `admin` not found)
    account_type=user
    account_id=admin
    correlation_id=8ca168ed8379438cbe6066d5cbcbe6dc
exit stat [... cut]
closed 2023-10-03 by nicholaspcr