TheThingsNetwork/lorawan-stack #6580: Support setting multiple frequency plans for gateways from the Console

KrishnaIyer, 2023-09-29:
### Summary

Support setting multiple frequency plans for gateways from the Console. 
The API supports setting `frequency_plan_ids` as an array and I think the console also uses this field but only sets one value.

### Current Situation

We can set only one frequency plan per gateway on the console.

### Why do we need this? Who uses it, and when?

Customer who use 16 channel (common) and 64 channel (uncommon) gateways cannot set additional frequency plans via the console. This is currently only supported via the CLI/API.

### Proposed Implementation

@kschiffer can you define the UX for this? I can imagine `+` button where users can "add" a new plan. Something like the attributes button below but then it opens up a dropdown instead of a text field.

[image]

Once the UX is defined, I'll triage/assign this for implementation.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

kschiffer, 2023-09-29:
Understood. Is the order of these FP settings of importance?

kschiffer, 2023-09-29:
[image]

@ryaplots can you pick this up?

KrishnaIyer, 2023-10-01:
> Understood. Is the order of these FP settings of importance?

Good question. For clients, I think we should call the API in the same order as the user sets it.

The general idea is that the order of setting should be the same order as the concentrator number. So fp[0] is set on concentrator [0] etc but I don't think this is documented or enforced.

However separately, @adriansmares/ @johanstokking the `io.Connection` stores the gateway frequency plans [as a map](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.28/pkg/gatewayserver/io/io.go#L112).
For example with LBS, the ordering of which FP generates which `SX1301_config` is not guaranteed.
If someone sets two difference FPs, then different FPs might be assigned to different concentrators at every reconnect. Is this something that we should look at preventing?

adriansmares, 2023-10-02:
> If someone sets two difference FPs, then different FPs might be assigned to different concentrators at every reconnect. Is this something that we should look at preventing?

The rules in general for having multiple frequency plans are quite loose, and the semantics are even less defined. In theory, yes the order matters and we should have a deterministic generation of the `SX1301_config`.

ryaplots, 2023-10-04:
Do we also want to display all the frequency plan ids for the gateway in the gateway overview in case it has several? Currently we display only one. @kschiffer @KrishnaIyer

PR #6615 (2023-10-19): Support setting multiple frequency plans for gateways from the Console

#### Summary

Closes #6580 

#### Changes

- Add gateway frequency plan select field component
- Update entry and key value map components to allow updating options
- Refactor fetch select
- Update validation schema
- Display the various frequency plan ids in gateway overview and in the general settings
- Fix bytes component (error in the console)

https://github.com/TheThingsNetwork/lorawan-stack/assets/56658938/a8b13f42-5bee-4460-9a9a-a6b417d910ef

#### Testing

1. Go to register a gateway.
2. Add a couple of frequency p [... cut]
closed 2023-10-19 by ryaplots