TheThingsNetwork/lorawan-stack #6598: TNN CLI runtime error when configuring (installed via snap)

Kostas-F, 2023-10-02:
### Summary

When trying to configure ttn-lw-cli with the EU1 regional cluster of the things stack community edition it results in a runtime error when generating the configuration file. In the step of setting identity server gRPC address in particular.

I installed via snap on xUbuntu 20.04.

When using the binaries from [here](https://github.com/TheThingsNetwork/lorawan-stack/releases) the configuration file is produced as expected.

### Steps to Reproduce

Run the following on ubuntu 20.04: 

1. `sudo snap install ttn-lw-stack`
2. `https://github.com/TheThingsNetwork/lorawan-stack/releases`
3. `https://github.com/TheThingsNetwork/lorawan-stack/releases`

### Current Result

Produces the runtime error as pasted in the relevant logs.

### Expected Result

The config file is produced.

### Relevant Logs

```shell
INFO	Telemetry is enabled. Check the documentation for more information on what is collected and how to disable it	{"documentation_url": "https://www.thethingsindustries.com/docs/reference/telemetry/cli"}
INFO	Configuring for The Things Stack Community Edition	{"cluster_id": "eu1"}
INFO	Set OAuth Server address	{"address": "https://eu1.cloud.thethings.network/oauth"}
INFO	Set Identity Server gRPC address	{"address": "eu1.cloud.thethings.network:8884"}
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x1d51edd]

goroutine 39 [running]:
go.thethings.network/lorawan-stack/v3/pkg/telemetry/exporter/cli.(*cliTask).Run(0xc000a5a000, {0x289a2c8?, 0xc000a12db0?})
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/telemetry/exporter/cli/cli.go:171 +0xfd
go.thethings.network/lorawan-stack/v3/cmd/ttn-lw-cli/commands.preRun.func1.2({0x289a2c8, 0xc000a12db0})
	/home/runner/work/lorawan-stack/lorawan-stack/cmd/ttn-lw-cli/commands/root.go:184 +0x145
created by go.thethings.network/lorawan-stack/v3/cmd/ttn-lw-cli/commands.preRun.func1
	/home/runner/work/lorawan-stack/lorawan-stack/cmd/ttn-lw-cli/commands/root.go:180 +0x574
```

### Deployment

The Things Stack Community Edition

### The Things Stack Version

3.27.2

### Client Name and Version

```shell
Using the cli ttn-lw-cli 3.27.2 on Ubuntu 20.04
```

### Other Information

The binaries as provided [here](https://github.com/TheThingsNetwork/lorawan-stack/releases) and specifically `lorawan-stack-cli_3.27.2_linux_amd64` did not result in the same error. 

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-10-03:
Thanks @Kostas-F for filing the issue. We are working on a fix for this.

Kostas-F, 2023-10-03:
> Thanks @Kostas-F for filing the issue. We are working on a fix for this.

No problem. If anymore info/testing that I could help with would be usefull let me know.

PR #6626 (2023-10-18): Fix snapcraft telemetry access bug

#### Summary

Closes #6598

Where snap application's permissions made a panic trigger when using the CLI provided in the snap deployment.

#### Changes

- Add `XDG_CACHE_HOME` to snapcraft options in goreleaser's release file.
- Add snapcraft options to goreleaser's snapshot file.

#### Testing

Install locally the snap and validate if the panic still happens.

<details>
<summary> Test steps </summary>

#### Requirements

If you are building the snap before the release you need to have the following:

- Be on a linux machine that has `snap` installed.
- Have [... cut]
closed 2023-10-18 by nicholaspcr