TheThingsNetwork/lorawan-stack #6655: segmentation violation on CLI connection

luciSolvis, 2023-10-25:
### Summary

I'm trying to connect to eu1.cloud.thethings.network but it returns a segmentation violation message.

### Steps to Reproduce

1. sudo snap install ttn-lw-stack
2. sudo snap alias ttn-lw-stack.ttn-lw-cli ttn-lw-cli
3. mkdir ttn
4. cd ttn
5. ttn-lw-cli use eu1.cloud.thethings.network

### Current Result

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x1d51edd]

goroutine 21 [running]:
go.thethings.network/lorawan-stack/v3/pkg/telemetry/exporter/cli.(*cliTask).Run(0xc00096c8d0, {0x289a2c8?, 0xc000c959e0?})
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/telemetry/exporter/cli/cli.go:171 +0xfd
go.thethings.network/lorawan-stack/v3/cmd/ttn-lw-cli/commands.preRun.func1.2({0x289a2c8, 0xc000c959e0})
	/home/runner/work/lorawan-stack/lorawan-stack/cmd/ttn-lw-cli/commands/root.go:184 +0x145
created by go.thethings.network/lorawan-stack/v3/cmd/ttn-lw-cli/commands.preRun.func1
	/home/runner/work/lorawan-stack/lorawan-stack/cmd/ttn-lw-cli/commands/root.go:180 +0x574

### Expected Result

create a file to describe the connection to eu1.cloud.thethings.network

### Deployment

The Things Stack Community Edition

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

nicholaspcr, 2023-10-26:
Hi @luciSolvis, from the result section this seems the same problem that we solved on `v3.28` ([referenced PR](https://github.com/TheThingsNetwork/lorawan-stack/pull/6626)). It is caused by the lack of permissions when attempting to create a file on snap deployments.

On Monday there should be the official release of `v3.28` with the fix but while that isn't available you can disable the telemetry by setting an environment variable.

```bash
export TTN_LW_TELEMETRY_ENABLE=false
```

Porl-F, 2023-10-27:
Thank you for posting the above. Very helpful just now.

nicholaspcr, 2023-10-31:
The release of `v3.28.0` on the [snapcraft](https://snapcraft.io/ttn-lw-stack) has been made and after testing it I confirmed that the bug is no longer present. 

I'm closing this issue since the problem has been resolved.

closed 2023-10-31 by nicholaspcr