TheThingsNetwork/lorawan-stack #6695: ABP Device Activation

RaysoftTechnologiesInc, 2023-11-09:
### Summary

I am currently looking to register a device via ABP, but the interface seem to have changed quite a bit and I cannot find a way to do so, could you please point me in the right direction because last time there was an Advanced option where I could choose between OTAA and ABP.

### Current Situation

Registering a device by ABP

### Desired Situation

To register a device by ABP

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-11-09:
Please see the documentation here https://www.thethingsindustries.com/docs/devices/adding-devices/#abp-devices

If something is missing there, you can open a issue to update the [documentation](https://github.com/TheThingsIndustries/lorawan-stack-docs/issues).

closed 2023-11-09 by KrishnaIyer

RaysoftTechnologiesInc, 2023-11-09:
Hi This is the interface I have and i cannot see what the documentation is
referring to

[image: image.png]

Where in that interface can i find the Advanced Activation?