TheThingsNetwork/lorawan-stack #6698: Generate CLI Configuration from the console

KrishnaIyer, 2023-11-09:
### Summary

Generate CLI Configuration from the console

### Current Situation

We currently go into the documentation and generate the first file. For subsequent clusters, we just make copies of the yaml file locally and replace the server address.

### Why do we need this? Who uses it, and when?

This would be great to quickly generate the `ttn-lw-cli.yml` config directly from the console. The console already has the required `base-domain` and the component addresses. 

### Proposed Implementation

- Have a button to generate a CLI config, maybe as part of a separate developer page.
- Option to disable telemetry (Sets `--telemetry.enable` to `false`).

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-11-09:
Phase 3 of https://github.com/TheThingsIndustries/product-management/issues/29

adriansmares, 2023-11-15:
We also have `ttn-lw-cli use` in order to generate the configuration file - is that slower than clicking the button in the Console and moving the file to the right location ?

KrishnaIyer, 2023-11-20:
That's a good first step for sure. This isn't well documented. I'll pick that up first.

closed 2024-10-06 by KrishnaIyer