TheThingsNetwork/lorawan-stack #6702: Missing API response for Webhook Downlink Scheduling

cwkhawand, 2023-11-12:
### Summary

Currently when POSTing to the Webhook Downlink API, the API response is empty.

### Steps to Reproduce

1. Make an API call to the Downlink Webhook API to send a downlink
2. Observe the response body

### Current Result

The response body is empty

### Expected Result

Ideally, the API should at least return the unique identifier of the downlink to be able to track it later when it is confirmed for example.

### Deployment

The Things Stack Community Edition

### Proposed Fix

Ideally the API should return the whole Downlink object data.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-11-16:
This happens via using correlation IDs. I'll add documentation on how to do this in the next few days.

cwkhawand, 2023-12-10:
Hello again,

Any possibility you could give me a hint on how to get the correlation ID after posting the downlink to TTN? At the moment the API returns an empty response so I'm not sure how I can retrieve the correlation ID.

KrishnaIyer, 2023-12-11:
Hey sorry for the delay. I'll add some quick docs this week on this.

closed 2025-04-30 by KrishnaIyer