TheThingsNetwork/lorawan-stack #6713: Gateways Server API: Bulk stats endpoint replies 404

dsepulvedar, 2023-11-17:
### Summary

I’ve been using the [Gateways Server API](https://www.thethingsindustries.com/docs/reference/api/gateway_server/#the-gs-service), specifically the bulk stats endpoint, to retrieve stats from my gateways.
Recently I hit a roadblock which defeats the purpose of the endpoint: when a gateway is not found, because it is offline and there’s no stats in the server, the endpoint replies with a `404` error response code, and says some of the gateways aren't found (see Current results)

As a next approach I decided to query the gateways' stats individually, and them remove the ones which were not found, and then query the bulk stats again with I thought was a list of gw that di have stats.
To my surprise, I still had the same `404` error response code, as seen below:
[image]

### Steps to Reproduce

1. Read gateways using the [/api/v3/search/gateways ](https://www.thethingsindustries.com/docs/reference/api/gateway/#the-entityregistrysearch-service) endpoint
2. Process the request and build a compatible payload for the [Gateways Server API](https://www.thethingsindustries.com/docs/reference/api/gateway_server/#the-gs-service)
3. Retrieve the stats and if there are gateways not found, the endpoint will reply with the `404` error response code.

### Current Result

This is response from the endpoint
```
{
 'code': 5,
 'message': 'error:pkg/identityserver:some_gateways_not_found (some gateways not found)',
 'details': [{
  '@type': 'type.googleapis.com/ttn.lorawan.v3.ErrorDetails',
  'namespace': 'pkg/identityserver',
  'name': 'some_gateways_not_found',
  'message_format': 'some gateways not found',
  'correlation_id': '1689af4e405f428e9059f8b02af7f602',
  'code': 5
 }]
}
```

### Expected Result

The endpoint should reply with the gateways' stats that are indeed found, and simply leave the devices that have been offline for more than 24 hours out from the response. 

### URL

https://<tenant>.nam1.cloud.thethings.industries/api/v3/gs/gateways/connection/stats

### Deployment

The Things Stack Cloud

### Client Name and Version

```shell
Python 3.11 with requests package
```

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-11-20:
This should be fixed by https://github.com/TheThingsNetwork/lorawan-stack/pull/6711 and will be part of the v3.28.1 release. The root cause seems to be that the caller (i.e. the admin user) is not a collaborator, and the batch rights assertion did not take into account the admin universal rights.

adriansmares, 2023-11-29:
@dsepulvedar do you still encounter this issue ?

dsepulvedar, 2023-12-05:
@adriansmares thank you for you heads up. I tested the endpoint again and I'm no longer getting the 404 "some_gateways_not_found" error. 
This to me has been solved.

closed 2023-12-05 by adriansmares