TheThingsNetwork/lorawan-stack #6748: Console event stream failure and closure behavior

adriansmares, 2023-12-04:
### Summary

Since the websocket refactor of the Console events streams, stream errors and closures seem to not be handled correctly. Specifically, the Console never reconnects the streams.

### Steps to Reproduce

Scenario 1:
1. Go [here](https://tti.staging1.cloud.thethings.industries/console/gateways/kerlink-ifemtocell-1).
2. Disconnect your WiFi / Ethernet cable.
3. After about 60 seconds, the websocket stream closes.

Scenario 2:
1. Locally, register a gateway and go to the live data page of the said gateway.
2. CTRL-C your locally run stack.
3. The websocket stream closes.

### Current Result

Scenario 1:
[image]
[image]
[image]

This is triggered _probably_ because errors from the `WebSocket` object are not actually instances of `Error` and as a result of this the [event conversion](https://github.com/TheThingsNetwork/lorawan-stack/blob/ecc4336084248e7341eb196d61f7af1c053e87e2/pkg/webui/console/lib/events/utils.js#L26) returns `undefined`. In turn, [this if](https://github.com/TheThingsNetwork/lorawan-stack/blob/ecc4336084248e7341eb196d61f7af1c053e87e2/pkg/webui/console/store/reducers/events.js#L56) errors out because it accesses a property of `undefined`.

Scenario 2:
Based on my debugging, the `close` event is triggered.

In both scenarios, the Console never automatically reconnects.

### Expected Result

The event streams should automatically reconnect on error/closure if these weren't client initiated.

### Deployment

The Things Stack Enterprise (self-hosted)

### The Things Stack Version

3.28.2

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

kschiffer, 2023-12-05:
I'll take care of this.

KrishnaIyer, 2023-12-19:
This is resolved right?

adriansmares, 2023-12-19:
Yes.

closed 2023-12-19 by adriansmares