TheThingsNetwork/lorawan-stack #6751: Improvement for notifications related to admin/tech contacts

nicholaspcr, 2023-12-06:
### Summary

From internal discussions the suggestion of not fanning out the notification to an `organization`'s collaborators. Instead, if an `organization` is specified as an `Admin/Tech` contact of another entity, we would propagate the notification to the `organization`'s respective `Admin/Tech` contact.

For example:
- `org-x` is the `admin_contact` of an application.
- `org-x` has `[usr-1,usr-2,usr-3]` as collaborators.
- `org-x` has `usr-Y` as its `admin_contact`

With the suggested changes instead of fanning out the administrative notification to `[usr-1,usr-2,usr-3]` we would send it to `usr-Y`.

### Current Situation

Currently when creating an notification we go through the `lookupNotificationReceivers` in order to know who are the receivers.

Flow of the method:
1. We fetch the `admin/tech` contacts from an entity and add it to the list of receivers.
https://github.com/TheThingsNetwork/lorawan-stack/blob/fe015cd59b6f5ffa9d42b5a200cdfaf58579f6ec/pkg/identityserver/notification_registry.go#L90-L130

2. Afterwards we check if any of the receivers is an `organization` and fetch its members as receivers:
https://github.com/TheThingsNetwork/lorawan-stack/blob/fe015cd59b6f5ffa9d42b5a200cdfaf58579f6ec/pkg/identityserver/notification_registry.go#L148-L160

So currently when an `organization` is an `admin/tech` contact we always fetch the members as the notification receivers.

### Why do we need this? Who uses it, and when?

The idea is to improve the notification system by avoiding excessive emails from being sent or try to find a way that its clearer for the user to organize its `admin/tech` contacts.

### Proposed Implementation

The initial idea is to replace the act of fetching the `organization`'s members and use its own `admin/tech` contact.

The problem here being that this disallows the (1:N) relationship that exists between a notification and its receivers.

So this issue needs a bit more of discussion regarding:
1. Is this a good improvement?
2. Is the way we create sets of notification receivers clear for the users?

cc: @adriansmares @KrishnaIyer 

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [ ] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-12-06:
So for the record even today it is possible to have a subset of the organization members receive the notifications, by creating a secondary organization only with the member subset, adding it as a collaborator, then setting it as a contact. It has the annoying ergonomics of having to add a secondary collaborator, which you can forget to update and maintain.

At the same time, it is equally annoying to have to fanout notifications to a 10 members organization, even if they don't want to receive them. Setting an individual user as a contact for these entities is problematic because replacing it would mean updating each individual entity.

I would propose that we add a field in the organization (`fanout_notifications`, a `BoolValue`, initially experimental) which controls this behavior - all of existing organizations should have it set to true while adding the column, for backwards compatibility, but for newer organizations I would leave this as false.

From there it should be easy - instead of `FindMembers` we retrieve the organization in order to check the `fanout_notifications`, `administrative_contact` and `technical_contact`, and based on `fanout_notifications` we either append only the contacts or every collaborator.

nicholaspcr, 2023-12-06:
> So for the record even today it is possible to have a subset of the organization members receive the notifications, by creating a secondary organization only with the member subset, adding it as a [...]

This is not possible because an admin has to be an organization's collaborator and nested organizations aren't allowed

nicholaspcr, 2023-12-06:
With my misunderstanding clarified I do agree with the idea of having the`fanout_notifications` field.

nicholaspcr, 2023-12-06:
With the improvement defined, it does require a migration but its quick work. 
So I believe that putting it on the `v3.29.0` milestone works fine.

KrishnaIyer, 2023-12-07:
I read through the issue and ACK with the design. Btw very good issue writing @nicholaspcr.

closed 2024-01-16 by nicholaspcr