TheThingsNetwork/lorawan-stack #6782: Revisit Safari WebSocket compression behavior

adriansmares, 2023-12-19:
### Summary

Currently Safari versions 15, 16 and 17 cannot properly operate when WebSocket compression is enabled. As such, compression is disable in the Console internal WebSocket API completely for Safari user agents. 

### Current Situation

Compression is completely disabled for Safari. We should recheck periodically (quarterly ?) if this ever gets fixed in Safari.

### Why do we need this? Who uses it, and when?

The bandwidth usage savings are non trivial when compression is enabled. At the same time, this problem exists since 2020, in many forms, so I am not particularly optimistic about a speedy fix.

### Proposed Implementation

Once the fix becomes widely adopted [on a market share basis](https://www.stetic.com/market-share/browser/), we can consider removing the Safari check completely.

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-12-20:
Ok let's put it into the backlog and revisit it every 3 months to see if Safari support is added.

KrishnaIyer, 2025-04-30:
No updates yet from Safari.

closed 2025-04-30 by KrishnaIyer