TheThingsNetwork/lorawan-stack #6805: Buf go sdk can't unmarshal timestamp

jpmeijers, 2024-01-03:
@jpmeijers we made some progress on this again: https://github.com/TheThingsNetwork/lorawan-stack/pull/6449.

We're now publishing clean protos to Buf: https://buf.build/thethingsnetwork/lorawan-stack. This allows you to generate code using `buf generate` for many languages, leveraging existing protoc plugins. See https://buf.build/docs/generate/usage/. There is no longer a dependency on this repository needed.

For Go, Node, Java and Swift, Buf also generates them on the fly, e.g. `go get buf.build/gen/go/thethingsnetwork/lorawan-stack/protocolbuffers/go@latest` and that's it (see https://buf.build/thethingsnetwork/lorawan-stack/assets/main).

_Originally posted by @johanstokking in https://github.com/TheThingsNetwork/lorawan-stack/issues/2297#issuecomment-1672554826_
            
I'm testing the Buf generated Go structs, but I'm getting an error when unmarshalling a JSON Application Uplink message:

`panic: json: cannot unmarshal string into Go struct field ApplicationUp.received_at of type timestamppb.Timestamp`

johanstokking, 2024-01-03:
This is because The Things Stack's JSON marshaling is not compliant with standard library `encoding/json` or even [`protojson`](https://pkg.go.dev/google.golang.org/protobuf/encoding/protojson). Early on we prioritized "nice" JSON over protobuf compliancy. One example is that lots of LoRaWAN stuff is hex encoded in JSON, while the proto field type is `bytes`.

When using webhooks, please use Protocol buffers as format.

johanstokking, 2024-01-03:
The alternative is to pull in TTS as dependency and use our JSON unmarshaling. But I think that's exactly what you're trying to avoid.

jpmeijers, 2024-01-04:
Due to historical reasons I have to support both JSON and Protocol Buffers. And I'd say that should also be the recommended way to write a webhook client for TTS.

Looking at the headers one can detect if it's JSON or Proto, and then split the handling to separate functions. The Proto handler can use Buf's definitions. Without having to include the whole TTS dependency, how would one generate the JSON handler? Using Swagger perhaps?

johanstokking, 2024-01-04:
If you'd only be interested in unmarshaling JSON without dependencies, don't bother with any proto at all (including Buf) and just declare your own structs with `json` tags. TTS V3 API applications facing is guaranteed to be stable.

> In my case I'll stick to importing TTS as dependency as long as it's possible, as that allows my to unmarshal both Proto and JSON to the same Go struct type. If I handle JSON and Proto with different handlers (different destination structs) I'll have to duplicate a lot of business logic.

In your case, as you support both protobuf and JSON, this is your only option.

closed 2024-01-04 by johanstokking