TheThingsNetwork/lorawan-stack #6824: Bug create/list device status OK but nthing in TTN Console

FK-sauve, 2024-01-10:
### Summary

"Hi all, this is my first contribution to TTN and GitHub, so it may not be very well-written. Apologies :) I created a Python script to create a 'Device' on TTN using the MQTT method. Everything works fine, but I don't see the devices created in the TTN console. Perhaps it's a bug, or there's certainly something wrong with my code."

### Steps to Reproduce

I publish detail and Code on my repo :                                        https://github.com/FK-sauve/BUG_TTN_V3_DEVICE_CREATE
1 You can use and adapt my python code by changing NWS key (MQTT Credentials)
2 Run  Python3 'create device TTN.py'
 2.1 status TTN OK (confirm device is created)
3 got ot TTN console and see there no device created
4 Run 'TEST_DEvice_TTN_EXIST.py'
 4.1 Status TTN OK (device exist)
5 You see  device created !? but nothing On TTN console

### Current Result

Device create and confirm

### Expected Result

see device on TTN console

### Relevant Logs

```shell
Please, See github for more details
```

### Deployment

The Things Stack Cloud

### The Things Stack Version

3.28.2

### Client Name and Version

```shell
Windows1O / Python3 / PAHO_MQTT
```

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-01-15:
Thanks for the issue but please include all the debugging steps in the issue template directly. We cannot unfortunately process an issue if the details are in a separate repository.

We will soon add clear documentation on how to create devices using the API. I'll link that here once done.

FK-sauve, 2024-01-20:
Hi M. Krishnalyer, woul you mind see more information below: # BUG_TTN_V3_DEVICE_CREATE
'''BUG-TTNV3
------------------- CREATE DEVICE RESULT ----------
runfile('A:/APP/root/UPPA-TEST/create device TTN.py', 
wdir='A:/APP/root/UPPA-TEST')

Dispositif eui-70b3d57ed0063abc approvisionné avec succès en mode ABP avec reset des compteurs.
{"ids":{"device_id":"eui-70b3d57ed0063abc","application_ids":{"application_id":"test-all"}},"created_at":"2024-01-10T05:05:56.196456285Z","updated_at":"2024-01-10T05:05:56.196456285Z"}
 status code: 200'''

Not seen in TTN console
 [image]
But Created In TTN:
[image]
 
'''--------------- TEST IF device EXIST -----------'''   
'''
runfile('A:/APP/root/UPPA-TEST/TEST_DEvice_TTN_EXIST.py', 
Return FOM TTN as exist:

wdir='A:/APP/root/UPPA-TEST')
Le dispositif eui-70b3d57ed0063abc existe dans l'application test-all.
Détails du dispositif:
{'ids': {'device_id': 'eui-70b3d57ed0063abc', 'application_ids': {'application_id': 'test-all'}}, 'created_at': '2024-01-10T05:05:56.196456285Z', 'updated_at': '2024-01-10T05:05:56.196456285Z'}
Caractéristiques du dispositif eui-70b3d57ed0063abc dans l'application test-all:
{'ids': {'device_id': 'eui-70b3d57ed0063abc', 'application_ids': {'application_id': 'test-all'}}, 'created_at': '2024-01-10T05:05:56.196456285Z', 'updated_at': '2024-01-10T05:05:56.196456285Z'}
---------------------------------------------------'''

---------------------------- PYTHON SCRIPT -----------------------------
1) Create:

# -*- coding: utf-8 -*-
"""
Created on Wed Jan 10 04:05:02 2024
This script demonstrate TTN device approvisionning
Python 3.11 Windows and linux work's well
@author: FSauve
"""

import requests
import json

# Credentials and all that we need to create
app_id = 'test-all'
access_key = '.......................4F4BUR4ANY.GFXCOLU22BOUARVUOOMPAE22INOOHTQBUWSTP2R3FFVPEI6GQJ2A'
dev_id = 'eui-70b3d57ed0063abd'
nwk_key = 'E5784B6171A5DAA2F1F5D8A6AB149ABD'
app_key = 'E8C54D73F451C89A1C402C49DF27DABD'

# URL de lAPI TTN V3 pour approvisionner un dispositif en ABP
# Mode ABP URL set in TTN DOC
api_url = f'https://eu1.cloud.thethings.network/api/v3/as/applications/{app_id}/devices/{dev_id}'

# Donnees du dispositif LoRaWAN
# Data to create device value to TTN
device_data = {
    "ids": {"device_id": dev_id},
    "lorawan_device": {
        "device_keys": {
            "nwk_key": nwk_key,
            "app_key": app_key
        },
        "supports_join": False,
        "mac_settings": {
            "reset_frame_counters": True
        }
    }
}

# En-têtes de la requête avec l'access key
headers = {
    'Authorization': f'Bearer {access_key}',
    'Content-Type': 'application/json'
}

# Envoi de la requête HTTP PUT pour approvisionner le dispositif
# Send HTTP request to TTN with credentials 
response = requests.put(api_url, headers=headers, data=json.dumps(device_data))

# Vérification du code de statut de la réponse
if response.status_code == 200:
    # response 200 = OK this device is create
    print(f"Dispositif {dev_id} approvisionné avec succès en mode ABP avec reset des compteurs.")
    print(response.text)
    print('--------------')
    print(' status code:', response.status_code)
else:
    # HTTP response != 200 so no creation
    print(f"Erreur lors de l'approvisionnement du dispositif. Code de statut : {response.status_code}")
    print(response.text)

-------------------------- TEST IF Device Create -----------------------------
# -*- coding: utf-8 -*-
"""
Created on Wed Jan 10 04:15:39 2024
Test if device exist in TTN
@author: FSauve
"""

import requests

# Remplacez ces valeurs par les informations spécifiques à votre application et votre dispositif
# app_id = 'votre_app_id'
# access_key = 'votre_access_key'
# dev_id = 'votre_dev_id'

app_id = 'test-all'
access_key = '....7EAFZ46SJEUSOAZVYMW4......GFXCOLU22BOUARVUOOMPAE22INOOHTQBUWSTP2R3FFVPEI6GQJ2A'
dev_id = 'eui-70b3d57ed0063abd'

# URL de l'API TTN V3 pour récupérer les détails du dispositif
# TTN url to testif device exist
api_url = f'https://eu1.cloud.thethings.network/api/v3/as/applications/{app_id}/devices/{dev_id}'

# En-têtes de la requête avec l'access key
headers = {
    'Authorization': f'Bearer {access_key}',
    'Content-Type': 'application/json'
}

# Envoi de la requête HTTP GET pour récupérer les détails du dispositif
response = requests.get(api_url, headers=headers)

# Vérification du code de statut de la réponse
if response.status_code == 200:
    # 200 = Exist
    print(f"Le dispositif {dev_id} existe dans l'application {app_id}.")
    device_details = response.json()
    print("Détails du dispositif:")
    print(device_details)
else:
    if response.status_code == 404:
        print(f"Le dispositif {dev_id} n'existe pas dans l'application {app_id}.")
    else:
        print(f"Erreur lors de la récupération des détails du dispositif. Code de statut : {response.status_code}")
        print(response.text)

# URL de l'API TTN V3 pour récupérer les détails du dispositif
api_url = f'https://eu1.cloud.thethings.network/api/v3/as/applications/{app_id}/devices/{dev_id}'

# En-têtes de la requête avec l'access key
headers = {
    'Authorization': f'Bearer {access_key}',
    'Content-Type': 'application/json'
}

# Envoi de la requête HTTP GET pour récupérer les détails du dispositif
response = requests.get(api_url, headers=headers)

# Vérification du code de statut de la réponse
if response.status_code == 200:
    device_details = response.json()
    print(f"Caractéristiques du dispositif {dev_id} dans l'application {app_id}:")
    print(device_details)

FK-sauve, 2024-03-04:
Hello, is there any issue about this bug? For information TTN says that device is created via API return status code 200, confirm device is really create but don't push the device in user's console. So the device remain a 'ghost' one and we can't delete it via ttn console. We can create a lot of device for debug purpose if log are require.Have a good day.

closed 2024-03-04 by FK-sauve