TheThingsNetwork/lorawan-stack #6946: Display both AFcntDown and NFcntDown on the device console

StevenCellist, 2024-02-21:
### Summary

Display the application downlink frame counter as well as the network downlink frame counter in the device console, next to the uplink frame counter.

### Current Situation

Currently, the single downlink frame counter only keeps track of the application downlink frame counter, and does not show anything related to the network frame counter. To find out the actual FCnt and keep track of network downlinks, one has to leave the console open to catch a network downlink, which shows the actual FCnt. Although, even this event only shows a complete FCnt, not a separation between AFcnt and NFcnt.

### Why do we need this? Who uses it, and when?

Adding the network downlink frame counter aids easier analysis of behaviour of individual nodes, e.g. performance under ADR. Some RF environments are quite unstable, causing many ADR downlinks; being able to compare these 'hidden' downlinks quantitively will help a lot. It also helps validating correct behaviour of nodes in general.

### Proposed Implementation

Change the current downlink frame counter such that it shows the AFcnt / NFcnt values when hovering over it with the cursor, or duplicate the downlink arrow, with e.g. one with N and one with A next to it, with a complete separation between the two.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [ ] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

StevenCellist, 2024-02-22:
No, sorry for not explaining clear enough. By that, I mean the following:
[image]
The frame counter at the top displays N/A (because I didn't send any downlink, i.e. AFcntDown = N/A), but the network has sent 143 downlinks, which I can only find by leaving the console open to catch these downlinks. Because in the JSON, it shows f_cnt = 142.
There currently is no way to see this real number of downlinks except for leaving the console open (for possibly, hopefully a very long time!) to catch one of these downlinks.

And sadly, the information in the JSON doesn't even discern AFcntDown and NFcntDown; in the next screenshot there is a device where I have sent a downlink through the console, but a network downlink reports the same f_cnt value as the full_f_cnt value.
[image]

I hope this explains what I would like to see improved!

ryaplots, 2024-02-29:
> Ok noted. This looks like a bug.
> 
> @ryaplots: What field is used by the Console to display the downlink frame counter?

The downlink counter in the title section is `last_a_f_cnt_down` - [https://github.com/TheThingsNetwork/lorawan-stack/blob/8449d7c93a111ee338a44db882d1da6f70f2457a/pkg/webui/console/store/reducers/devices.js#L103-L110](https://github.com/TheThingsNetwork/lorawan-stack/blob/8449d7c93a111ee338a44db882d1da6f70f2457a/pkg/webui/console/store/reducers/devices.js#L103-L110)

[https://github.com/TheThingsNetwork/lorawan-stack/blob/8449d7c93a111ee338a44db882d1da6f70f2457a/pkg/webui/console/store/reducers/devices.js#L177-L188](https://github.com/TheThingsNetwork/lorawan-stack/blob/8449d7c93a111ee338a44db882d1da6f70f2457a/pkg/webui/console/store/reducers/devices.js#L177-L188)

StevenCellist, 2024-03-06:
@KrishnaIyer that would be very much appreciated! My ideas would be a) two downlink counters side by side, or b) showing the separate counters upon mouse hover.

kschiffer, 2024-03-07:
Yeah let's store and display both downlink counters:

[image]

StevenCellist, 2024-03-07:
@kschiffer yes perfect! The only thing I wonder, is if this is clear enough for new users, as to what they should look at, but I myself would be very happy with it.

kschiffer, 2024-03-07:
Yeah, we'll add a detailed description to the tooltip (help icon) as well.

ryaplots, 2024-04-05:
What should be the text of this detailed description? @KrishnaIyer @kschiffer

KrishnaIyer, 2024-04-05:
I think this following makes sense:

`Nwk`: frame counter for network downlinks (`FPort` = 0)
`App`: frame counter for application downlinks (`FPort` >=1)

closed 2024-04-08 by ryaplots

StevenCellist, 2024-05-03:
@kschiffer @ryaplots first and foremost thank you very much for implementing the separate counters: makes my life much easier and looks great!
I'm sorry for bumping this; however, it seems the implementation is not yet perfect. I'd like to show two cases:
* When no (App) downlink is sent, the arrow and `n/a` text is missing - see the following screenshot
[image]
There _is_ some sort of `n/a` going on, which I think would be for the Nwk counter, as this following device has no downlinks at all:
[image]
It just doesn't appear to be working well for the App downlinks.

* When a new downlink occurs while the Console is open (at least for Network downlink; didn't test Application downlink yet as otherwise the above screenshot does not occur), the frame counters do weird things - see the following screenshot
[image]
The AFcntDown value takes the new value of NFcntDown, while the previous value stays the same... this may have something to do with the `f_cnt` / `full_f_cnt` field not showing any distinction between App/Nwk downlinks, but I am not aware of the details on these fields.

And in some case, which I don't know how to reproduce yet (screenshot from a couple days ago), the counters get the same value even though the App should still be at `n/a`:
[image]
This is likely in the same ballpark as the previous point.

If I can help by sharing access to these LW v1.1 devices (or maybe you can already access them, dunno), let me know please! Would be happy to test :)

PR #7022 (2024-04-08): Display both AFcntDown and NFcntDown in the Device overview in the Console

#### Summary

Closes #6946 

#### Changes

- Store both counters in the state
- Show both counters when available
- Show `help` icon for tooltip

[image]

#### Testing

1. Log into the Console
2. Go to the overview of an end device with LW 1.1.x
3. See two downlink counters if both the network and the app have downlinks and only one of the counters of only the app/network has downlinks.
4. See also a description about this in the tooltip

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unre [... cut]