TheThingsNetwork/lorawan-stack #6956: Console icon library

pierrephz, 2024-02-28:
### Summary

Find the right icon library

### Current Situation

Currently, for the console redesign, we chose to use [Google Material Symbols](https://fonts.google.com/icons?icon.set=Material+Symbols&icon.style=Rounded).

For some reasons, we face issues with some icons, for example the _star_ icon and the _plus_ icon. They look a bit off on development.

### Why do we need this? Who uses it, and when?

Enhance and smooth UX, give visual cues and add personality to the console

### Proposed Implementation

We can look for another icon library: first find consistent, trustable, professional look icon library. And then test them.

I can do researches, submit a list of icon libraries that would match with our product. Then we should agree on 1 or 2, then test them.

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

pierrephz, 2024-02-28:
Here is a selection of Icon library that could potentially match our product:

[Font Awsome](https://fontawesome.com/search?s=regular&f=classic&o=r)
- 3281 icons
- Paid

[Mingcute](https://www.mingcute.com/)
- 2810 icons
- Open source

[Tabler icons](https://tabler.io/icons)
- 4962 icons
- Open source

[Iconic](https://iconic.app/)
- 1328 icons
- Paid

those are 4 libraries that I think could match, especially Tabler icons. I can find more if we don't agree on one of them

cc @kschiffer

kschiffer, 2024-03-05:
We decided on the frontend meeting today to use Tabler icons.

KrishnaIyer, 2024-04-11:
Since #6996 is merged, we can close this.

PR #6996 (2024-03-28): Introduce new icon set
#### Summary

This PR introduces the new icon set based on Tabler icons, and replaces the old material icon font.

Closes #6956 

#### Changes

- Add tabler icon package (react binding)
- Remove material icon font
- Refactor codebase to update used icons
- Apply a couple of fixes that result from using SVG icons instead of icon fonts

#### Testing

It's hard to test all possible icons in the GUI, so random sampling should be enough.

##### Regressions

It's possible to some undefined icons are remaining. However the icons have been replaced by a small script that I wrote [... cut]
closed 2024-04-11 by KrishnaIyer