TheThingsNetwork/lorawan-stack #6966: Support fine-grained NbTrans controls while using Dynamic ADR mode

KrishnaIyer, 2024-03-04:
### Summary

Support fine-grained NbTrans controls while using Dynamic ADR mode. Ref: https://thethingsindustries.atlassian.net/servicedesk/customer/portal/1/TTS-6383

### Current Situation

In the Dynamic ADR mode, we currently cannot set a max NbTrans for specific data rates.

### Why do we need this? Who uses it, and when?

For customers who need this control without switching to the Static mode.

### Proposed Implementation

Already done in https://github.com/TheThingsNetwork/lorawan-stack/pull/6962. This issue is filed retroactively.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-03-04:
This has two parts
- [ ] Backend implementation; https://github.com/TheThingsNetwork/lorawan-stack/pull/6962
- [ ] Console support

##### Console Support

In the `Device General Settings` window, under `Advanced MAC settings`.
- If the `Adaptive Data Rate (ADR)` is set to `Dynamic Mode`
- We can set upto 16 Data Rate "options". `Data Rate 0` to `Data Rate 15`.
- Each data rate can have one set of Min NbTrans and Max NbTrans
- Validation; 1 <= `Min NbTrans` <= 15,  1 <= `Max NbTrans` <= 15 and `Min NbTrans` < `Max NbTrans`
[image]

cc: @kschiffer for UX.

KrishnaIyer, 2024-03-06:
The API/backend part is done. @kschiffer please check the UX for this and @ryaplots can pick this up for `v3.30.0`.

kschiffer, 2024-03-07:
[image]

Here's the wireframe for the functionality. This one assumes that there's also a default value that can be set, which I believe should be possible as well. cc @KrishnaIyer

KrishnaIyer, 2024-03-11:
The API defines both to be between `1` and `3`.
So I can say the default `Min. NbTrans` is `1` and the `Max. NbTrans` is `3`.

ryaplots, 2024-03-12:
I have a question about this:
1. @kschiffer @KrishnaIyer  The "Default for all data rates" fields are not possible to change, right? The only real user action here is adding overrides, correct? Or is there API to add default data rates?
2. @kschiffer I am using the key value map component that I edited to add a selector at the beginning because this component has all the perfect logic to deduce the field values of `overrides`. However, it does not have the field titles per key/value field and adding them seems like extending the component too much and changing its purpose. Can we use placeholders instead of field titles like below?
[image]

kschiffer, 2024-03-13:
Thank you! The screenshot looks good already.

1. So apparently the default can not be changed but is fixed to `1` and `3`. I believe it's good to show that as well. Let's replace the two fields with a text. `The network server applies a min NbTrans of 1 and a max NbTrans of 3 for all data rates. You can use the button below to define overrides.`
2. It should be relatively easy to achieve the same thing without `<KeyValueMap />` by using form hooks and rendering normal `<Form.Field />`'s based on some internal state or even the form values directly. Ideally, we would be able to solve these things also with the key-value-map but it would need a refactor to make better use of the form hooks and allow for objects as values.

adriansmares, 2024-03-13:
To clarify the API a bit:
1. It is possible to not have per data rate overrides, and to have a default (i.e. used without an override):
https://github.com/TheThingsNetwork/lorawan-stack/blob/e9e18b674116416dc9493da3065ce51e0933416f/api/ttn/lorawan/v3/end_device.proto#L449-L460
2. The range is indeed always 1 to 3 at most.

kschiffer, 2024-03-13:
Ah so in that case, my initial wireframe is actually more suitable.

So it would be possible to have a (modified) default and a couple of overrides, right?

adriansmares, 2024-03-13:
> So it would be possible to have a (modified) default and a couple of overrides, right?

Yes. In general I wouldn't show / set these unless someone explicitly requested these overrides, but it is possible.

kschiffer, 2024-03-14:
After some discussion on Slack, the new layout is as follows:
[image]

Text for the `Use default settings for number of retransmissions` checkbox tooltip:
> The number of retransmissions (NbTrans) controls how many times a frame will be transmitted over the air. The redundancy introduced by retransmissions improves the chances that a packet will be received, at the expense of more power usage. By default, depending on the number of missed frames, the same frame may be transmitted 3 times.

Text for the `Data rate specific overrides` tooltip:
> Data rate specific overrides allow the number of transmissions to be limited on a per data rate basis. This may be used to limit power usage for low data rates.

Also note the following:
The data rates in the dropdown should be resolved to the actual symbol like `SF7BW125`. These can be fetched via:
https://github.com/TheThingsNetwork/lorawan-stack/blob/89515596adecaf13548737c8e065819a41d0f7ca/api/configuration_services.proto#L173

@adriansmares how to correlate the bands with the data rate index? Couldn't find that out on a quick glance.

adriansmares, 2024-03-14:
For an individual end device, you know the frequency plan ID and regional parameters version. You already have all of the frequency plans in memory (for rendering purposes), so based on that you can lookup the band ID in the frequency plan (similar to how you look at the names currently, but `.band_id` instead of `.name`).

The extra step is to download all of the band definitions (via that RPC). Let's call the result of that `descriptions`. `descriptions[bandID].band[regionalParametersVersion]` is the band definition, and the data rate descriptions are in `.data_rates`. `.data_rates[i]` is the definition of data rate index `i`. You can then translate this definition to text similar to how we render `SF7BW125` in the events view, and you are done.

PR #7012 (2024-04-10): Support NbTrans controls while using Dynamic ADR mode

#### Summary

Closes #6966 

#### Changes

- Add store for band definitions
- Add NbTrans override fields (default override and specific override)
- Add encoding and decoding of the checkbox
- Fix browser console errors

#### Testing

https://github.com/TheThingsNetwork/lorawan-stack/assets/56658938/3defb1ed-9673-4c1b-bcf6-2f9cccd7c90e

1. Login into the console
2. Go to any device/create and application and device
3. Go to device settings, to Network Server section and open the collapsed "Avanced MAC settings" section.
4. Select ADR Dynam [... cut]
closed 2024-04-10 by ryaplots