TheThingsNetwork/lorawan-stack #6981: Show JoinEUI and DevEUI for activation in live traffic view

johanstokking, 2024-03-11:
### Summary

Show the JoinEUI And DevEUI for activation in live traffic view.

### Current Situation

Currently, we show the (new) DevAddr. The EUIs are only shown for some events:

[image]

### Why do we need this? Who uses it, and when?

During activation, what's more relatable are the EUIs as they are stable. The new DevAddr is useful too, but not as useful as it is presented now as the only identifier.

### Proposed Implementation

I think this is frontend only as these identifiers should be available in the events already.

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

kschiffer, 2024-03-13:
I have no other suggestions that adding the DevEUI to the event preview, whenever it is available.

johanstokking, 2024-03-14:
> I have no other suggestions than adding the DevEUI to the event preview, whenever it is available.

I think it is always available, same as the JoinEUI, for every device related event as it is part of the primary identifiers.

I'm not against always showing the DevEUI, but it's going to take a lot of space. Not sure though.

I would like to see the JoinEUI + DevEUI at least for all activation-related events.

kschiffer, 2024-03-14:
Sorry, this was actually a typo. I meant `JoinEUI`.