TheThingsNetwork/lorawan-stack #7006: LoRaWAN v1.1 ABP: SNwkSIntKey incorrectly listed as NwkSKey on console Overview

StevenCellist, 2024-03-25:
### Summary

For LoRaWAN v1.1 devices registered in ABP mode, the SNwkSIntKey name in the device Overview tabs appears to be incorrect.
As shown in [this scheme](https://www.thethingsnetwork.org/docs/lorawan/end-device-activation/#activation-by-personalisation-in-lorawan-11), there should be SNwkSIntKey, FNwkSIntKey, NwkSEncKey and AppSKey. These are correctly listed on the tab General settings, under Network layer and Application layer. However, the Overview tab shows the NwkSKey instead of the SNwkSIntKey. The value of NwkSKey listed does align with the SNwkSIntKey under Network layer.

### Steps to Reproduce

1. Register device as LW v1.1 in ABP mode
2. Open Overview tab
3. Open General settings > Network layer
4. Compare values and names of keys

### Current Result

SNwkSIntKey is listed as NwkSKey

### Expected Result

NwkSKey should be renamed to SNwkSIntKey

### Deployment

The Things Stack Community Edition

### The Things Stack Version

3.29.1

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-03-26:
[image]

[image]

> NwkSKey should be renamed to SNwkSIntKey

What is the basis for that? 

`NwkSKey` value is not defined in LW1.1.x.  In LW1.0.x, the `NwkSKey` is responsible for both encryption and integrity (MIC checks) whereas in LW 1.1.x, these are split into three separate keys ( `SNwkSIntKey`,  `NwkSEncKey` and`FNwkSIntKey` with the latter related to roaming).

The `NwkSKey` value that's being displayed on the console for LW 1.1.x is actually the `FNwkSIntKey`, which is also technically incorrect.

I think what we should do is to make the  keys in the _session information_ section of overview tab consistent with what's shown in the `Network Layer`, `Application Layer` and `Join Settings` sections of the `General Settings` page.
 - For LW 1.1, replace `NwkSKey` label with `FNwkSIntKey` and keep the rest the same.
 - For LW 1.0, display only the `NwkSKey` and `AppSKey` in the session information tab.

cc: @kschiffer @johanstokking

StevenCellist, 2024-03-26:
Oops, my bad - had been looking at the keys so hard trying to figure out what was up with them that I mixed up the Serving and Forwarding key.. sorry!

descartes, 2024-03-26:
Ha, I knew this would happen but I've learnt not to test the Gods of ~~Olympus~~ LoRaWAN!

@StevenCellist's been making RadioLib LW stack soooo much better & I'm his test bunny who went snow-blind trying to configure ABP over the weekend, so I take some of the blame.

closed 2024-04-08 by ryaplots

StevenCellist, 2024-04-29:
@KrishnaIyer thank you for following up! I am closely tracking the release and already prepared the changes to match the newly correct keys 😃  (https://github.com/jgromes/RadioLib/commit/6591e983140754b129a1e34b681fc3a69d478032)

PR #7020 (2024-04-08): Fix session information

#### Summary

Closes #7006 

#### Changes

- Replace NwkSKey label with FNwkSIntKey for LW 1.1.x 
- Display only the NwkSKey and AppSKey for LW 1.0.x

[image]

[image]

#### Testing

1. Log into the console.
2. Open the overview of a device with a LW 1.1.x
3. In the Session information you should see the FNwkSIntKey
4. Open the overview of a device with a LW 1.0.x
5. In the Session information you see only only the NwkSKey and AppSKey.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibilit