TheThingsNetwork/lorawan-stack #7063: Do not autogenerate/prefill gateway ID in the console

KrishnaIyer, 2024-05-02:
### Summary

Do not autogenerate/prefill gateway ID in the console.

### Current Situation

If a gateway has an EUI, the console prefills the ID field as `eui-<EUI>`. The intention of this was to make things easier for the users.

### Why do we need this? Who uses it, and when?

In reality, however, this is causing more pain and support questions in The Things Stack Sandbox. This is the cause for the most asked question on the forum
1. Users add a gateway EUI.
2. The console prefills the ID.
3. User's select the FP and just click yes.
4. This gateway is deleted (usually to be handed out to someone else). The ID is not purged since the user is usually not an admin.
5. The new owner of the gateway registers it. The console prefills the ID. But now since the ID is not purged, they get an error and contact the forum.

Despite documentation and multiple forum posts, users still repeat this issue. It's time for us to fix this via UX.

### Proposed Implementation

1. Don't auto-fill the ID based on the EUI. Make users choose this. It's only one extra step.
2. Improve the tooltip for this field to explain that the ID is not the EUI and this is chosen by the user.
[image]

3. On deleting the gateway, add a note that the ID is locked until purged by an admin but the EUI can be reregistered later with a different ID.

Any thoughts @kschiffer?

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2024-05-03:
This should also apply to device IDs.

kschiffer, 2024-06-25:
I don't know how proportionate it is to remove a convenience for 95% of users because 5% are having issues. I'd rather like to find a way to mitigate the issue while keeping the convenience.

Is there any way we can find out up front whether a ID is taken, without being an admin?

Also, it's likely possible to improve the guidance in the gateway onboarding for the case that the gateway ID is already taken by handholding users more to simply use another ID.

kschiffer, 2024-06-25:
[image]
I believe this error message is the real issue.

This should be caught in the frontend and replaced with
> The gateway ID is already in use in this network, please choose another one.

KrishnaIyer, 2024-06-26:
> Is there any way we can find out up front whether a ID is taken, without being an admin?

This is indeed one option but we don't have an RPC for this and we'd need to consider the security implications of allowing non-admins to just do a search on IDs.

> I don't know how proportionate it is to remove a convenience for 95% of users because 5% are having issues. I'd rather like to find a way to mitigate the issue while keeping the convenience.

I think that we are causing failures by making things too convenient. Users just inattentively click the next field without being made to think what they are filling. 
The ID is the primary identifier to reference the gateway across multiple data sources. I think that users should consciously pick a memorable value rather than auto-generating `eui-{EUI}`.

descartes, 2024-06-27:
> I think that we are causing failures by making things too convenient. Users just inattentively click the next field without being made to think what they are filling.
> The ID is the primary [...]

This, this is the heart of the matter - gateway lifetime is in years, you accumulate them and then have to figure out what's where - JeffUK has his own encoding scheme that tells him at a glance what make/model/location a gateway is.

A few extra seconds setup saves hours in all sorts of ways when debugging devices.

There is one other wrinkle in this that causes problem - gateways and the global_conf.json references "gateway_ID" which is actually the EUI as we all know - but just occasionally we see people entering the Id from the console in to the gateway setup. Doesn't come up often, but as it's on the same layout ...

kschiffer, 2024-06-27:
I see. Thanks for the additional insight. Then let's indeed enforce conscious input for IDs.

KrishnaIyer, 2024-08-22:
This is tested and I can confirm that we don't prefill the IDs.

closed 2024-08-22 by KrishnaIyer