TheThingsNetwork/lorawan-stack #7093: Normal users without admin rights can still create apps, gateways

kewe1, 2024-05-21:
### Summary

On my private TTS environment iothink-solutions.eu1.cloud.thethings.industries. I need to invite some users with limited access and rights that corresponds to their organization.

For some reason, despite having set as a non admin and having those option checked, the user can still create his own application and gateways : 
[image]
[image]

### Steps to Reproduce

1 invite a new user in user management as approved without admin rights
2 create an organization with roles limiting the user to only see and interact on his organization app
3 connect on this new user
3 the new user can create a new application outside his organization

### Current Result

The user shouldn't be authorized to create apps and gateways outside his organization without admin rights

### Expected Result

The user can only interact with his application from his organization. Since he is not an admin, he cannot create an app or gateway outside of his organization possible rights.

### URL

iothink-solutions.eu1.cloud.thethings.industries

### Deployment

The Things Stack Cloud

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Validation

- [ ] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-05-27:
Hey @kewe1. Changes to tenancy settings take at least 10 mins to reflect due to database caching. Can you try now again?

nicholaspcr, 2024-05-27:
Hello @kewe1, this happens because the tenant configuration is cached. 
The changes have an affect after 10 minutes, so after this time window the non admin users should not be able to create the referenced entities.

Let me know if this is not the case.

kewe1, 2024-05-27:
Hello @nicholaspcr,

Indeed, when I tested, it was within that 10minute frame, I can confirm that I cannot create applications now as a normal user. 
Good to know and sorry for creating an issue that was just a cache configuration timing!

Thanks!

closed 2024-05-27 by kewe1