TheThingsNetwork/lorawan-stack #7098: Cannot add application error:pkg/auth/rights:no_user_rights

tanawat369, 2024-05-23:
### Summary

Hello 
i try to add application but found error

### Steps to Reproduce

i send a request to 
http://192.168.10.100/api/v3/users/rvp-admin1/applications
{
  "application": {
    "ids": {
      "application_id": "test123"
    },
    "name": "My Test Application",
    "description": "Application for my test devices",
    "network_server_address": "thethings.example.com",
    "application_server_address": "thethings.example.com",
    "join_server_address": "thethings.example.com"
  }
}

### Current Result

response
{
    "code": 7,
    "message": "error:pkg/auth/rights:no_user_rights (no rights for user `rvp-admin1`)",
    "details": [
        {
            "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
            "namespace": "pkg/auth/rights",
            "name": "no_user_rights",
            "message_format": "no rights for user `{uid}`",
            "attributes": {
                "uid": "rvp-admin1"
            },
            "correlation_id": "6130c2dbc87f41dab6675b13ff97aa7d",
            "code": 7
        }
    ]
}

[image]

### Expected Result

i want to add application

### Deployment

The Things Stack Enterprise (self-hosted)

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Validation

- [ ] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-05-23:
> i send a request to
http://192.168.10.100/api/v3/users/rvp-admin1/applications

How are you sending the request? Via the CLI or API? How are you setting authentication for the call?

tanawat369, 2024-05-24:
> > i send a request to
> > http://192.168.10.100/api/v3/users/rvp-admin1/applications
> 
> How are you sending the request? Via the CLI or API? How are you setting authentication for the call?

> > i send a request to
> > http://192.168.10.100/api/v3/users/rvp-admin1/applications
> 
> How are you sending the request? Via the CLI or API? How are you setting authentication for the call?

test via postman and use bearer authen

Tertoey, 2024-05-28:
> Ok and what rights are you assigning to the API key?

Grant all current and future rights
now i can add application via path api/v3/organizations/rvp-admin1/applications

KrishnaIyer, 2024-05-28:
Ok so the issue is resolved?

KrishnaIyer, 2024-07-30:
Closing the issue due to a lack of input.

closed 2024-07-30 by KrishnaIyer