TheThingsNetwork/lorawan-stack #7150: Deleted user can't be restored or purged

kewe1, 2024-06-20:
### Summary

I have an user on my TTS account that I cannot restore or purge, it shows an empty error message on the down right notification

I also tried adding it back normally and it does nothing. 

User account is : je-bouazzaoui

### Steps to Reproduce

1. Access admin panel
2. Acess user management
3. Deleted (Admin)
4. Click on restore
5. Errors pops

### Current Result

I cannot restore back my user

### Expected Result

I want my user "je-bouazzaoui" restored

### URL

iothink-solutions.eu1.cloud.thethings.industries

### Deployment

The Things Stack Enterprise (self-hosted)

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Validation

- [X] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-06-20:
This is a valid bug. @ryaplots the console makes a DELETE request to `/api/v3/users/{users.user_id}` instead of `/api/v3/users/{users.user_id}/purge` when purging a user.

KrishnaIyer, 2024-07-29:
This is completed via https://github.com/TheThingsNetwork/lorawan-stack/pull/7164 and internally tested.

closed 2024-07-29 by KrishnaIyer