TheThingsNetwork/lorawan-stack #7223: Add an option to pause webhooks

KrishnaIyer, 2024-08-06:
### Summary

Add an option to pause webhooks.

### Current Situation

Currently, the only way to stop a webhook is to delete it (or if it's unhealthy and we set a threshold to stop it). 

### Why do we need this? Who uses it, and when?

This can be very useful for testing where users can pause webhooks while the external server is being updated without needing to delete them.

### Proposed Implementation

I'll get back here when we get this into a milestone.

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

vlasebian, 2024-09-12:
Tested on staging1 as follows:

Prerequisites:
- Create an app on staging1
- Connect a generic node to the app
- Login the CLI tool to the staging environment 

Steps:
- Create a webhook in the console, use a webhook site to see the requests (e.g webhook.site)
  - Enable uplinks for the webhook
  - Enable downlink acks for the webhook
After the webhook is created, requests are going to be sent to the webhook site every time an uplink is sent.
- Pause the webhook using the cli tool:
```bash
-> % go run ./cmd/ttn-lw-cli --config .env/configs/staging1.yml app webhooks get vv-play wh1
WARN	No fields selected, will select everything
{
  "ids": {
    "application_ids": {
      "application_id": "vv-play"
    },
    "webhook_id": "wh1"
  },
  "created_at": "2024-09-12T08:09:47.311512669Z",
  "updated_at": "2024-09-12T08:11:35.323370856Z",
  "base_url": "https://webhook-test.com/87adc5fb8ac89205940ec603dbb218b9",
  "format": "json",
  "uplink_message": {},
  "health_status": {
    "healthy": {}
  },
  "field_mask": {
    "paths": []
  }
}

-> % go run ./cmd/ttn-lw-cli --config .env/configs/staging1.yml app webhooks set vv-play wh1 --paused
{
  "ids": {
    "application_ids": {
      "application_id": "vv-play"
    },
    "webhook_id": "wh1"
  },
  "created_at": "2024-09-12T08:09:47.311512669Z",
  "updated_at": "2024-09-12T08:12:17.238958811Z",
  "base_url": "https://webhook-test.com/87adc5fb8ac89205940ec603dbb218b9",
  "format": "json",
  "paused": true
}
```
- Requests to the webhook site are stopped
- Enable the webhook again
- Requests to the webhook site are resumed

There is one issue with the CLI tool flag. The CLI tool can only set the paused flag of the webhook to true, it doesn't accept the `--paused false` option. Will look into it. To resume the webhook I used an API request:
```bash
curl -X PUT "https://tti.staging1.cloud.thethings.industries/api/v3/as/webhooks/$APP/$WEBHOOK" \
  -H "Authorization: $TTN_STAGING_AUTH_TOKEN" \
  -d '{
  "webhook": {
    "ids":{ "webhook_id": "'"$WEBHOOK"'" },
    "paused": '"$PAUSED"',
    "field_mask":{"paths":[]},
    "format":"json"
  },
  "field_mask": {
    "paths": [
      "ids.webhook_id",
      "paused",
      "field_mask",
      "format"
      ]
    }
  }'
```
Response:
```json
{"ids":{"application_ids":{"application_id":"vv-play"},"webhook_id":"wh1"},"created_at":"2024-09-12T08:09:47.311512669Z","updated_at":"2024-09-12T08:16:03.259567478Z","base_url":"https://webhook-test.com/87adc5fb8ac89205940ec603dbb218b9","format":"json","field_mask":{"paths":[]}}
```

KrishnaIyer, 2024-09-12:
Ok thanks. This is great. I'll add a note here for UX and console support.

pierrephz, 2024-10-16:
## Here are some wireframe and user flow to pause / run a webhook:

1. On the webhook list: hover the webhook you want and you will see a `Pause` button
2. A dialog will pop to ask you to confirm `Pause` (since this action can cause unwanted effects)
3. A new `Paused status` is set in the `Status column`
4. When hovering this webhook again, the button will now be a `Run` button

### Wireframes:

[image]
[image]
[image]
[image]

KrishnaIyer, 2024-10-17:
@mjamescompton for the implementation.

KrishnaIyer, 2024-12-02:
This is tested on staging and works pretty well.

closed 2024-12-02 by KrishnaIyer