TheThingsNetwork/lorawan-stack #7308: Support user notification preferences

KrishnaIyer, 2024-09-16:
### Summary

Support user notification preferences. Replaces https://github.com/TheThingsIndustries/lorawan-stack/issues/3280

### Current Situation

Currently users receive emails for all notification types except `collaborator_changed`. We want to allow users to choose what kind of emails they receive.
All notifications are displayed in the console. So some users prefer to not receive emails

### Why do we need this? Who uses it, and when?

We want to allow users to choose what kind of emails they receive.

### Proposed Implementation

#### Backend

- [ ] Introduce a new `Enum` for `NotificationType` in the API. Currently we use hardcoded strings like `user_requested` or `collaborator_changed` in `pkg/email/templates` and when we use the `CreateNotificationRequest` message.
   - [ ] We need to do a replacement in all the places where this is used and also test that email templates are not affected by this.
- [ ] Introduce a `NotificationPreferences` message inside the [User](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.32/api/ttn/lorawan/v3/user.proto#L291)  message.
   - [ ] This contains a single message `repeated NotificationType email_notification_types`. This is a list of `NotificationType` for which we would send emails.
   -  [ ] This can be updated by the User based to select which notifications they want emails for. We need some validation to filter out duplicates.
   - [ ] There are special admin emails `user_requested` and `client_requested`, both of which require admin actions. These emails are very infrequent and should not be skipped. So the API should either ignore these enums or return a validation error.
- [ ] Add DB entries and migrations. 
- [ ] Change the [lookupNotificationReceivers](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.32/pkg/identityserver/notification_registry.go#L127) function to return a second `emailReceiverIDs` argument. This is a filtered list of receiver IDs.
  - [ ] A receiver is placed in this list if the email notification type in `req` is allowed for the user in their preferences.
- [ ]  In [`CreateNotification`](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.32/pkg/identityserver/notification_registry.go#L254), call `is.SendNotificationEmailToUserIDs` only `emailReceiverIDs` is not empty.
- [ ] Deprecate the `email` field of the [Notification](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.32/api/ttn/lorawan/v3/notification_service.proto#L60) message. We should instead get this from the user preferences
 - [ ] Lots of unit and production testing

#### UX

- [ ] Edit the `/console/user-settings/profile` view and rename it to `Account` or `Preferences`, whichever makes sense.
- [ ] Add a new "Email Notifications" section.
  - [ ] This contains a list of checkboxes, similar to choosing Rights on API keys. Example below
    [image]
  - [ ] The options are the `NotificationType` enums.
  - [ ] We also somehow have an option to choose `All` or `None`.
  - [ ] Let's document here that emails that require admin actions such as approving users or OAuth clients cannot be skipped.
- [ ] Plug this page into the API above and test.

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-09-16:
@nicholaspcr: Please let me know if the backend part makes sense?

@pierrephz: Can you think of a draft wireframes for the UX? Please ping me if you have doubts.

nicholaspcr, 2024-09-16:
> @nicholaspcr: Please let me know if the backend part makes sense?

Looks pretty good to me but I do have a question, would it be more efficient to have a `NotificationsFilters` instead of `NotificationsPreferences`? That way what we are storing on the user are only the notification types which have to be filtered.

Additionally I believe its easier for users to set what they don't want rather than the opposite.

KrishnaIyer, 2024-09-17:
> Looks pretty good to me but I do have a question, would it be more efficient to have a NotificationsFilters instead of NotificationsPreferences? That way what we are storing on the user are only the notification types which have to be filtered.

The question here is the difference of what the default behavior is, 
a. Do users receive all emails by default and they choose what not to receive? (OR)
b. Do users receive no emails by default and they can enable it for certain ones?

User experience-wise, (b) is better since all notifications are available in the Console by default.

ryaplots, 2024-09-19:
@KrishnaIyer @nicholaspcr @pierrephz Considering that `user_requested` and `client_requested` require admin actions, they should be selected by default and the de-selection should be disabled in the UI. There should also be a message there saying that these require user action and that they are infrequent, so they cannot be opted out of. What do you think?
For the backend, I am not sure if it's best to ignore these or return an error. I would say error. In case one of these gets through, then it's nice to have some kind of feedback. Have them as "not allowed" options.

KrishnaIyer, 2024-09-20:
> Considering that user_requested and client_requested require admin actions, they should be selected by default and the de-selection should be disabled in the UI. 

I would say let's show it in a non-editable state in the console for admins only. For the API, I think we should just ignore this if it's set in the filters.

ryaplots, 2024-10-07:
@KrishnaIyer In the `notifyAdminsInternal()` [method](https://github.com/TheThingsNetwork/lorawan-stack/blob/5978c07f9ab925f55896b8420132503770cfe391/pkg/identityserver/notification_registry.go#L291), can we just ignore the `req.Email` field that will be deprecated and the allowed notification types as well? This method is only used for `user_requested` and `client_requested` types and these notifications should always send emails to admins, so we don't need to check if the user allows the notification types.

KrishnaIyer, 2024-10-08:
> @KrishnaIyer In the notifyAdminsInternal() [method](https://github.com/TheThingsNetwork/lorawan-stack/blob/5978c07f9ab925f55896b8420132503770cfe391/pkg/identityserver/notification_registry.go#L291), [...]

Yes I think that makes sense.

pierrephz, 2024-10-15:
Email notification wireframe:

1. Access the `Email notification settings` in the side bar under `User settings`
2. Find the list of preferences (For admin, some are disabled if they require their actions)
3. After editing, save or discard and receive a feedback from the system

We can still discuss the list of preferences, for example, I am not sure that `Temporary password` should be an option there as you will always receive an email for that.

[image]

ryaplots, 2024-10-29:
I am not sure the wireframe is correct. Do we need options for notifications like validate or login token? Also the descriptions don't seem correct as an API key can be changed for different entities and not only apps. @KrishnaIyer can you double check?

I have opened a draft PR with fixed descriptions and included only the necessary notifications - [https://github.com/TheThingsNetwork/lorawan-stack/pull/7363](https://github.com/TheThingsNetwork/lorawan-stack/pull/7363)

cc: @KrishnaIyer @pierrephz  let me know if it's good

KrishnaIyer, 2024-10-29:
@ryaplots: These are just examples. You can adapt it to the list of options that we actually provide. If you need help in drafting text for each of the options, let me know.

KrishnaIyer, 2024-12-02:
I did a basic test on `staging1` and this looks good. We can come back here if we find issues with the RFT.

closed 2024-12-02 by KrishnaIyer