TheThingsNetwork/lorawan-stack #7318: Join server dev nonce metrics reversed

onizmx, 2024-09-24:
### Summary

Not sure if this has any production implication, but happened to step on observability.go and discovered:
registerDevNonceReuse is using `jsMetrics.devNonce.tooSmall` and
registerDevNonceTooSmall is using `jsMetrics.devNonce.reuse`

### Steps to Reproduce

https://github.com/TheThingsNetwork/lorawan-stack/blob/94ea5fbe3f433624aa8a3e58dc6a925ac8ffd89f/pkg/joinserver/observability.go#L132

### Current Result

Not sure, perhaps metrics are reversed?

### Expected Result

registerDevNonceReuse should be `jsMetrics.devNonce.reuse` and
registerDevNonceTooSmall shoud be `jsMetrics.devNonce.tooSmall`

### Deployment

The Things Stack Open Source (self-hosted)

### The Things Stack Version

3.28.0 but also checked the latest release

### Proposed Fix

registerDevNonceReuse should be `jsMetrics.devNonce.reuse` and
registerDevNonceTooSmall shoud be `jsMetrics.devNonce.tooSmall`

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Validation

- [ ] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-10-10:
@ryaplots: Looks like this is a valid request to me but can you check if this is correct? If yes, the fix would be quite simple.

KrishnaIyer, 2024-10-17:
Thanks for raising this @onizmx. This is fixed and will be live with the `v3.32.2` release scheduled at the end of this month.

closed 2024-10-17 by KrishnaIyer