TheThingsNetwork/lorawan-stack #7328: Upgrade React to Version 18

kschiffer, 2024-10-08:
### Summary

Upgrade the current React version to 18 for improved performance and new features.

### Current Situation

We are currently using a version of React prior to 18. This version lacks features such as automatic batching and concurrent rendering, which are available in React 18. These enhancements are crucial to improve the performance and responsiveness of our UI.

### Why do we need this? Who uses it, and when?

Developers will benefit from this upgrade, particularly when building more interactive and performance-heavy components. Users of our application will experience improved responsiveness and performance, especially in UI components that involve complex state changes.

### Proposed Implementation

- Upgrade the `react` and `react-dom` packages to version 18 in package.json.
- Update any dependencies that rely on React to ensure compatibility.
- Test the application to confirm no breaking changes occur, especially with third-party libraries.
- Follow the [official React 18 upgrade guide](https://react.dev/blog/2022/03/08/react-18-upgrade-guide) to ensure the smooth transition of existing features.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [ ] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

kschiffer, 2024-10-08:
Blocking #7329

KrishnaIyer, 2024-10-10:
@mjamescompton and @PavelJankoski: Can you guys investigate this and make a proposal of the steps necessary for this?
We can then plan this accordingly.