TheThingsNetwork/lorawan-stack #7346: Live data split view interaction

pierrephz, 2024-10-17:
### Summary

Improve the live data split view interaction.

### Current Situation

## Problem:

On a end device or gateway screen, we have a feature that allows user to always keep an eye on the live data of that device.
But the floating button to access it is not visible enough and people don't notice it.

[image]

### Why do we need this? Who uses it, and when?

This feature is used to be able to do debugging, and make some editing on devices and always have a look on the live data. It make life of the user easier since he or she doesn't have to switch page many time to check if the device receives data.

### Proposed Implementation

## Solution:

1 - Improve the floating button visibility (see screenshot below)

[image]

2 - Show an explanation tooltip when the user first access this page (see screenshot below)

[image]

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

pierrephz, 2024-10-17:
@KrishnaIyer here is a solution to make the `live data split view` more accessible.

KrishnaIyer, 2024-10-17:
This looks great and will definitely improve this view. Thanks

mjamescompton, 2024-10-17:
@KrishnaIyer Lets think about where we would store the data that the explanation tooltip has been seen, if its just a cookie it will be annoying ask it will open for every browser. 

Do we have something like last login date that could be used to hide and show this.

Do we have somewhere we can store this small bits of data for ui elements

KrishnaIyer, 2024-10-21:
Yeah this setting should be in the [UserConsolePreferences](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.32/api/ttn/lorawan/v3/user.proto#L58).

@ryaplots: Can you add the backend support for this?

KrishnaIyer, 2025-02-03:
Tested in the RFT

closed 2025-02-03 by KrishnaIyer