TheThingsNetwork/lorawan-stack #7347: Dark mode

pierrephz, 2024-10-17:
### Summary

Introduce the dark mode feature.

### Current Situation

Currently we only have a light mode.

### Why do we need this? Who uses it, and when?

Like in a lot of products and web apps, we should be able to switch themes to increase accessibility.

### Proposed Implementation

Here is a wireframe:

1 - Add a `Theme` setting under `User settings`
2 - Choose between `Light`, `Dark` and `System` mode
3 - Save or Discard

[image]

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-11-12:
This looks great and ok for me. As discussed in the tech sync, this will require some update to the tokens in the process.

pierrephz, 2024-11-29:
here are the assets to be able to build the settings page:

[image]

[image]

[image]

----------

Download .zip
[console-theme-settings-assets.zip](https://github.com/user-attachments/files/17961390/console-theme-settings-assets.zip)

KrishnaIyer, 2025-03-26:
linked to https://github.com/TheThingsNetwork/lorawan-stack/pull/7529

closed 2025-04-14 by KrishnaIyer