TheThingsNetwork/lorawan-stack #7348: Move the Quick actions' panel to the top

pierrephz, 2024-10-17:
### Summary

Move the Quick actions' panel to the top.

### Current Situation

Based on some feedbacks, when you first log into the console and don't have any application, gateways or end devices, this is pretty hard to know how to get started since you have to scroll to the bottom of the dashboard to find the Quick actions' panel.

### Why do we need this? Who uses it, and when?

Facilitate the onboarding of new users, or fasten the process of adding application or device in the console for regular users.

### Proposed Implementation

Mockup with reworked quick actions' panel on the top of the dashboard for TTS Cloud.

[image]

Mockup with reworked quick actions' panel on the top of the dashboard for TTS Discovery.

[image]

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-10-17:
@pierrephz: Since we have this in the nav bar already, can we remove this panel and instead update the UX of this item to make it more prominent than the `+` sign?
[image]

pierrephz, 2024-10-17:
This is not a problem to have two or more ways of adding an item in the console.

The Dashboard is here to be interactive and give as many generic options as we can, including taking actions. IMO we should not remove this quick actions' panel.

About the `+` button in the header, I agree and upgrading it was also in my mind; find below a mockup

[image]

KrishnaIyer, 2025-02-03:
Tested in the RFT

closed 2025-02-03 by KrishnaIyer