TheThingsNetwork/lorawan-stack #7380: Implement MAC settings profiles

halimi, 2024-11-12:
### Summary

Introduce end device MAC setting profiles that contain MAC settings that can be applied to multiple end devices.

### Current Situation

Currently if you want to change the MAC settings for a batch of end devices then you need to do an individual call for each end device.

### Why do we need this? Who uses it, and when?

To able to update easily a batch of end devices with the same MAC settings. It is useful if you have a big fleet of end devices.

### Proposed Implementation

1. Create a new API endpoint to create/read/update/delete MAC settings profile
2. Modify the end device creation API to able to set a MAC settings profile ID to an end device
3. Create a new API where you can set the MAC settings profile ID for batch of devices in the application
4. Modify NS to use the MAC settings profile if it set
   - Nullify the original settings and use the settings from the profile
5. Modify NS to save the settings from the profile to the end device settings when the profile is disassociated
6. Create a new API endpoint to create MAC settings profile from an existing end device MAC settings

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-11-12:
Supersedes https://github.com/TheThingsIndustries/product-management/issues/44

halimi, 2025-11-10:
This has been implemented and deployed to production.