TheThingsNetwork/lorawan-stack #7382: Normalized message pre processor not using the json schema

peter-quandify, 2024-11-15:
### Summary

The json schema for the normalized uplink is located in the device-library. It is currently expanding but the message preprocessor is located in the lorawan-stack and does not use the schema supplied in the device-library for validation. Instead it uses a custom go code to validate.

This causes the validators to be out of sync and currently the stack validator is missing quite a few fields.

If only validation is needed in this function, then a library like this could do the trick
https://github.com/kaptinlin/jsonschema

### Steps to Reproduce

1. Add a device with a normalized uplink decoder that has the battery or metering field
2. Check the log output on decoding to see that it fails

### Current Result

Generates an error for each field it does not recognize

### Expected Result

Should accept everything that the device-library schema accepts

### Deployment

The Things Stack Community Edition

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Validation

- [ ] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2024-11-18:
@peter-quandify thanks for reporting, we're fixing this.

KrishnaIyer, 2024-12-02:
This is deployed to all clusters and the warnings are gone. Please feel free to comment if not solved and we'll take a look.

peter-quandify, 2024-12-02:
Still get these errors on the TTN cloud console
[image]
[image]

johanstokking, 2024-12-02:
@happyRip

KrishnaIyer, 2024-12-04:
@peter-quandify: Which TTSS (TTN) Cluster are you testing on?

peter-quandify, 2024-12-06:
@KrishnaIyer I'm on EU1, the error is still present when I click on Live data in the application. Not on live data for that specific device.

[image]

KrishnaIyer, 2024-12-09:
@peter-quandify: we've deployed an update to The Things Stack Sandbox `eu1` cluster. Let us know if this is fixed.

KrishnaIyer, 2024-12-18:
We have made yet another update and we don't see these warning logs on our side. Can you check?

KrishnaIyer, 2025-04-30:
@peter-quandify: Please let us know if this is not fixed.