TheThingsNetwork/lorawan-stack #7441: Custom LNS URI is not sent to the gateways over CUPS

BenoitDuffez, 2024-12-10:
### Summary

It seems that it is possible to change the LNS address to another one, for example on port 443 instead of the default 8887.
https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.33/pkg/basicstation/cups/config.go#L36

Then, CUPS will send this address to the gateway but if it is empty or has no port, it will fallback to 8887, which won't work.
https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.33/pkg/basicstation/cups/server.go#L225

### Current Situation

This makes a hard requirement for each gateway to be configured with the appropriate server address, including the port.

### Desired Situation

I think it would be nice to infer the default port from the custom LNS URI instead of the hard-coded 8887.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2024-12-11:
We use the hardcoded value of `8887` only if either the gateway specific LNS URI or the default LNS URI do not contain a port https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.33/pkg/basicstation/cups/server.go#L205

So I'm not totally sure I understand the issue here? Can you show an example of what you're expecting and what you're getting?

BenoitDuffez, 2024-12-11:
Thanks. I have a customer that has (I believe) an on prem install on `xxx.customer.com`. They have configured the LNS to listen on `443` because it offers better compatibility with SIM cards and exotic internet providers (who may block `8887`).

Then when we provision the gateways, if we leave the gateways server empty or without a port, it will fall back to the default `8887`.

I had assumed that CUPS would _know_ that the LNS is not listening on `8887` but whatever the service is started on (in our case, `443`) and send this to the gateway over CUPS.

Now the simple fix is simply to declare the gateways in the backend with an explicit LNS that includes the port, however if there is only one LNS under that CUPS, I had assumed that it would infer directly what the actual port is instead of a hard-coded one.

Does that make more sense?

KrishnaIyer, 2024-12-11:
Ok I see. In The Things Stack, CUPS and LNS are implemented by two different micro services that don't really know where each other is hosted.
If you register a gateway without setting the `gateway-server-address` field and if you set `--gcs.basic-station.default.lns-uri` to the `443` endpoint, this will tell CUPS to use the default value.

BenoitDuffez, 2024-12-11:
I had assumed indeed that they would be two different services, but I had hoped that they would know about each other or that the default CUPS port could be overridden the way the LNS port can be (I don't know how the CUPS service is started, but it would be nice if it had a flag `--lns-uri xxx:443` for example).

Anyway I got my answer, if you won't change this implementation you may close this ticket.

Thanks!

KrishnaIyer, 2024-12-11:
Great. Thanks!