TheThingsNetwork/lorawan-stack #7466: Cannot remove all device attributes in UI

vlasebian, 2025-01-07:
### Summary

I cannot remove all the device attributes in the device settings.

### Steps to Reproduce

Prerequisites:
- Have TTS OS running locally
- Have an application already created

Steps:
- Go to TTS home page
- Register an end device
- Go to the settings of the end device (Applications -> App Name -> End devices -> Dev name -> Settings)
- Add a few attributes
- Save the changes
- Go to another page and then come back on the device settings page
- Remove all the device attributes by clicking on the remove button in the right of each attribute

### Current Result

The 'Attribute values must have less than 200 characters' error is shown and changes cannot be saved. If you navigate to another page and come back, the attributes are still there.

[image]

### Expected Result

No error should be shown. End device attributes should be removed when clicking on save changes.

### Deployment

The Things Stack Open Source (self-hosted)

### Client Name and Version

```shell
The Things Stack for LoRaWAN: ttn-lw-stack
Version:             3.33.0-dev
Go version:          go1.23.3
OS/Arch:             darwin/arm64
```

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Validation

- [ ] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2025-01-13:
I can confirm this one. Probably a front-end validation issue. @PavelJankoski can you take a look?

KrishnaIyer, 2025-02-19:
This was already mereted to `v3.33.1`.

closed 2025-02-19 by KrishnaIyer