TheThingsNetwork/lorawan-stack #7468: Devices from other application with similar device-id prefix shown in sidebar's top end devices

StevenCellist, 2025-01-10:
### Summary

I have access to two applications, both with devices that have their device-id starting with `mapquester`. Similar to #7321, these devices show up in the other application, but apparently also in the sidebar which I didn't notice before.

### Steps to Reproduce

1. Create two applications
2. Register devices in both that have an identical prefix for their device-ids
3. Check the top end devices once they appear there

### Current Result

There are only two devices in this application, but the sidebar on the left lists four more (R/G/B/001a) that are from a similar but other application. Not _actually_ a leak because I have access to both either way, but not very much supposed to happen :)
[image]

### Expected Result

Don't show the devices :)

### Deployment

The Things Stack Community Edition

### The Things Stack Version

v3.33.0.3c8e962fdd

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Validation

- [ ] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2025-01-13:
Hmm I don't think we use a prefix to filter the device IDs, we use the application identifier. @ryaplots can you see if we can reproduce this?

StevenCellist, 2025-01-13:
FYI: the application ID for both applications also use the same prefix 😅 (`mapquester` vs `mapquester-v1`).

ryaplots, 2025-01-13:
Yeah so we indeed don't use prefixs to fill either device IDs or application IDs. I took a quick look and couldn't reproduce this. But what happens in the screenshot is indeed wrong, because only the top devices of the currently open applications should be displayed in the applications layer. So I will look more into it

StevenCellist, 2025-01-30:
Tested OK, thanks all!

[image]

closed 2025-01-30 by StevenCellist