TheThingsNetwork/lorawan-stack #7498: Password manager no longer auto fills

virtualguy, 2025-02-04:
### Summary

Password manager (1password) no longer detects the password field on the login page

### Steps to Reproduce

Browse to login page with password manager extension installed

### Current Result

No prompt for password to be filled

### Expected Result

Password auto fills

### Relevant Logs

```shell

```

### Deployment

The Things Stack Enterprise (self-hosted)

### Client Name and Version

```shell

```

### Other Information

This looks to have been intentionally changed by @kschiffer  here:  
https://github.com/TheThingsNetwork/lorawan-stack/blob/b253e54d9971bf054ef837bab0c571abc047c013/pkg/webui/components/input/index.js#L222

### Contributing

- [ ] I can help by doing more research.
- [x] I can help by implementing a fix after the proposal above is approved.
- [x] I can help by testing the fix before it's released.

### Validation

- [ ] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [x] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

kschiffer, 2025-02-04:
Some context: this change was added to prevent inputs for sensitive data such as JoinEUIs from showing the password manager overlays, if this also prevents password inputs from showing the overlays (even though they don't explicitly set the `sensitive` prop) then that is indeed a bug that needs to be fixed.

KrishnaIyer, 2025-02-04:
@PavelJankoski: Can you take a look? 

@virtualguy Kevin does not work with us anymore. For future issues, please don't tag members directly. Github automatically adds the `needs/triage` label for all new issues and we have an internal process to triage them.

johanstokking, 2025-02-04:
I think we need to split the definition of sensitive then: passwords should be filled, but root keys shouldn't, for example.

KrishnaIyer, 2025-04-30:
This is fixed in #7500