TheThingsNetwork/lorawan-stack #7522: Device attributes not returned via API in The Things Stack Self-Hosted v3.33.0

matiasAS, 2025-02-28:
### Summary

I am using The Things Stack Self-Hosted v3.33.0, and I have noticed that device attributes are not returned via the API, even though the documentation suggests they should be accessible.

### Steps to Reproduce

1. I have devices with attributes configured in the TTS Console.
2. I attempted to retrieve them using the following API call:

   ```sh
   curl -X GET "https://<my-tts-server>/api/v3/applications/<application_id>/devices/<device_id>" \
     -H "Authorization: Bearer <my-api-key>" \
     -H "Content-Type: application/json" \
     -d '{
       "field_mask": {
         "paths": ["attributes"]
       }
     }'

### Current Result

The response does not include the attributes field.

### Expected Result

The API should return the attributes field as per the documentation.

### Relevant Logs

```shell

```

### Deployment

The Things Stack Enterprise (self-hosted)

### The Things Stack Version

3.33.0

### Client Name and Version

```shell
Using the CLI: ttn-lw-cli 3.33.0 (run `ttn-lw-cli version`) on Ubuntu 22.04 (Intel)  
Using a web browser: Firefox 102.0 on Ubuntu 22.04 (Intel)  
Gateway: Milesight UG65 Semi-Industrial with the latest firmware  
TTS Deployment: Self-hosted using Docker (`thethingsnetwork/lorawan-stack:3.33.0`)
```

### Other Information

TTS Version: v3.33.0 Self-Hosted
Database Check: The attributes do exist in the PostgreSQL database (attributes table).
API Key Permissions: The API key has full rights.
TTS Logs: No error messages related to attributes retrieval.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Validation

- [x] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [x] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

techminion940118, 2025-02-28:
Hello Matias, I can help you.

matiasAS, 2025-02-28:
Ok @techminion940118 , do you know if it is possible? Can we talk here or another way (whatsapp, email, telegram...)?
I'm from Chile, I don't speak English, but with the translator I can!

techminion940118, 2025-02-28:
Okay, no problem. Can you share your email, so that I can invite you in my slack workstation?

matiasAS, 2025-02-28:
matialvarezs@gmail.com

matiasAS, 2025-02-28:
@techminion940118 where you from?

techminion940118, 2025-02-28:
I am from Canada, I have just sent invitation.
https://join.slack.com/t/docker-assistant/shared_invite/zt-30wkfan0k-62TYZOgnMuHXnoBP6OMTTA

matiasAS, 2025-03-03:
@KrishnaIyer thanks

[image]

Do you know the solution to that error? It is on a proxmox server, pfsense is my firewall on a virtual machine, I have ports 80/443 busy, I am using external letsencrypt ssl and I pass them to docker by volume, and it gives me that error.

KrishnaIyer, 2025-03-04:
For this please check our troubleshooting: https://www.thethingsindustries.com/docs/enterprise/docker/troubleshooting/.

matiasAS, 2025-03-06:
Hello @KrishnaIyer ,

I followed the troubleshooting guide ([link](https://www.thethingsindustries.com/docs/enterprise/docker/troubleshooting/)) but ACME is not generating the <domain>+rsa certificate inside my Docker container.

Setup:
Proxmox VM with pfSense firewall
Ports 80 and 443 are forwarded to the TTS VM
TTS runs in Docker, listening on 80 and 443
ACME volume: ./acme:/var/lib/acme
Issue:
The TTS stack starts successfully, but Let's Encrypt challenges return 404 at http://tts1.sistemasmas.cl/.well-known/acme-challenge/test.
Port 80 is open and listening (netstat -tulnp).
No ACME logs in docker logs the-things-stack-2-stack-1 --tail=50 | grep acme.
The <domain>+rsa file is not generated in ./acme/.
I tried removing ./acme/ and restarting, but the issue persists.
Questions:
Why is ACME not generating the <domain>+rsa certificate?
How can I manually force TTS to request a new certificate?
Can I reach you via WhatsApp, Slack, Skype, or another channel for direct assistance?
Thanks in advance!

Best regards,
Matias

KrishnaIyer, 2025-03-10:
For these questions, please check our forum: https://www.thethingsnetwork.org/forum/

Since the original topic in the issue doesn't seem to be an issue, I'm closing this.

closed 2025-03-10 by KrishnaIyer