TheThingsNetwork/lorawan-stack #7531: ACME certificate not generated correctly in Docker environment (Self-hosted)

matiasAS, 2025-03-06:
### Summary

I am running The Things Stack in a self-hosted environment with ACME for certificate generation. However, the ACME process does not generate the expected +rsa file, causing login failures due to invalid certificates.

### Steps to Reproduce

1- Set up TTS in a Proxmox VM behind pfSense, with ports 80 and 443 forwarded.
2- Deploy using Docker Compose with the ACME configuration:
tls:
  source: acme
  acme:
    dir: "/var/lib/acme"
    email: "<email>"
    hosts: ["<domain>"]
    default-host: "<domain>"

4- Run docker compose up -d
5- ACME does not generate the expected +rsa certificate.
6 -Login fails with "token exchange refused" error.

### Current Result

No valid certificate is generated.
Login to the Console fails with "token exchange refused".
OpenSSL verification fails with unable to get local issuer certificate.

### Expected Result

ACME should generate a valid certificate.
Login should work without errors.

### Relevant Logs

```shell

```

### Deployment

The Things Stack Enterprise (self-hosted)

### The Things Stack Version

3.33.1

### Client Name and Version

```shell
Proxmox 8.3.4
TTS Version: 3.33.1
Pfsense CE as firewall of 

Virtual machine OS: Ubuntu 22.04

Docker Compose Version: 1.29.2
Self-hosted (not The Things Stack Cloud)
```

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Validation

- [ ] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [x] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

matiasAS, 2025-03-10:
Hello @KrishnaIyer ,

On my server with proxmox, virtual machine, production server, then I think that self-signed ssl is not ideal. Also because what I deploy works without problem on the first try on an ec2, vps contabo or hetzner, but on my server that I have to do port forwarding it gives me that error in the image. The big difference: port forwarding and that in the outgoing rules I allow everything except 22
The error is:

[image]

matiasAS, 2025-03-10:
Hello @KrishnaIyer ,

If I deploy on an ec2, contabo or hetzner vps it works without problem on the first try, but on my server with proxmox that I have to do port forwarding I get the error in the image. The big difference: port forwarding and that in the outgoing rules I allow everything except 22

The error is:
[image]

Best regards,
Matias

KrishnaIyer, 2025-03-11:
The Things Stack is agnostic of the environment you're deploying this in. You say that this works on EC2 for example. The difference between that and proxmax is outside the scope of The Things Stack and that's for you to investigate.

closed 2025-03-11 by KrishnaIyer

matiasAS, 2025-03-11:
@KrishnaIyer 
Is it compatible if port forwarding is done? Because it is something that I must investigate myself? Do you 100% rule out that it is something in the compatibility of port forwarding with the auth system? Will it be compatible with dns-01?