TheThingsNetwork/lorawan-stack #7543: Remove/Move Live Data button

ElcompMatt, 2025-03-11:
### Summary

Currently in the V3 cloud console (v3.34.0.1e2cb40100), there is a live data button which obscures the last activity of the last row of the **End Devices** list in an application. 

We are in the very fortunate situation to now have applications with many devices in them and this is more of an annoyance than anything.

The image below is showing the issue: 

[image]

### Current Situation

We often have to grab screen shots to send to site engineers during installs and the last device time is important for the commissioning process and evidence pack to show the devices communicating. We currently have to cut copy paste, which isn't an issue in itself, it just looks a bit shabby. 

Please bear in mind, that the people assembling the commissioning pack are office administrators. 

### Why do we need this? Who uses it, and when?

Despite the obvious reason, which is a button should not cover or obscure data, it does detract from how functional the site is. It is also a little frustrating when you are trying to do performance checks, when you just want to see the last comms time and you have to sort, copy paste etc.

### Proposed Implementation

What would be useful, assuming it doesn't already exist, is the ability to move or turn it off completely in the settings. The live data functionality is already available in the nav bar menu options, so this does seem superfluous, and the live data pop up drawer (in my opinion only) is not an enjoyable user experience, especially when trying view the payload.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [x] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [x] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2025-03-12:
Thanks @ElcompMatt: We'll check this. @pierrephz can you check the UX here please? I think this request is valid.

KrishnaIyer, 2025-04-14:
The latest release adds a padding at the bottom so that the last items at the bottom are not obscured.

closed 2025-04-30 by KrishnaIyer